Frix-x/klippain #3: Cura help

PhilBaz, 2022-04-25:
Im having trouble getting Cura to recognize & use the script.

https://gist.github.com/frankbags/c85d37d9faff7bce67b6d18ec4e716ff

 I'm placing it in C:\Users\xxx\AppData\Roaming\cura\master\plugins\, as instructed here : https://github.com/Frix-x/klipper-voron-V2/blob/main/macros/probing/bed_mesh.cfg .

I've tried on 5.0 Beta, as well as 4.13 master fork. 

Any help would be appreciated.

Frix-x, 2022-04-27:
Hello, sorry for the delay !

I'm not using Cura myself but just tried and get it working by putting the script manualy in this directory:
`C:\Program Files\Ultimaker Cura 5.0.0\share\cura\plugins\PostProcessingPlugin\scripts`

I think it's because its more like a hack and not a real Cura plugin and there was probably some changes in the last versions of the software... I'll update my small documentation with that, thanks for the hint !

closed 2022-04-30 by Frix-x

PhilBaz, 2022-05-01:
Thank you!!! Appreciate it. Adaptive bed mash is the best!!!