Frix-x/klippain #10: Naming inconsistency vibr_calibrate_02.cfg

MartinGiersberg, 2022-08-20:
I belive there are some slight inconsistancyies in the Instructions of vibr_calibrate_02.cfg

#   4. Add vibrations_graph.py in the same folder in your config. You can find this file in my scripts folder
#   4bis. Make it executable using SSH: type 'chmod +x ./vibrations_graph.py' when in the folder.

The provided python script in [graph_vibrations.py](https://github.com/Frix-x/klipper-voron-V2/blob/main/scripts] is however called graph_vibrations.py, so just the two words spun arround.

Thanks a lot for you work !!!

Frix-x, 2022-08-20:
Hey, thank you very much for this higlight ! I'll correct that mistake :)

MartinGiersberg, 2022-08-20:
You are more than welcome ! Thanks again for developing this tool!

commit 7076ecce (2022-08-20): updated vibr_calibrate_02 documentation (fix #10)
closed 2022-08-20 by MartinGiersberg