Frix-x/klippain #14: add move type to enter probe dock when attaching

elpopo-eng, 2022-08-25:
In the current version, when we want to attach a probe, we perform the entry movement reversing the exit movement.
For euclid it's better to enter by the side the movement is softly

So I propose to add a type/lenght of movement in config 

```
variable_probe_move_to_enter_when_attaching: "front"
variable_probe_move_to_enter_when_attaching_length: 20
```

Frix-x, 2022-08-26:
I wanted to keep it simple and used the attach move in reverse for the dock entry. I need to think about a better way to do it without adding a new "move" that will be in reverse than the two others... and a lit bit hard to understand at first. Maybe setting an entry point with coordinates is necessary for this one... ?

Maybe something like: `variable_probe_before_entry_position: xxx,yyy`

What do you think about that ?

elpopo-eng, 2022-08-26:
I'm also not very happy with setting a position, because in some cases we may not have the same entry when docking and attaching.

In fact we do not define the movement in the config but a position alias after the movement.
If we just rename variables it should be simpler :
```
variable_probe_before_attach_postion : 'back'
variable_probe_after_attach_postion : 'right'
variable_probe_before_dock_postion : 'right' // generaly same as probe_after_attach_postion
variable_probe_after_dock_postion : 'back'
```

In addition I was troubled by the front and back movement, for me it's the opposite (Ymax is the back of the machine for me)

```
#    Y
#    ^ 
#    |         back
#    |           ^
#    |   left  < O >  right
#    |           v
#    |          front
#    |_ _ _ _ _ _ _ _ _ _ _ _> X
```

Frix-x, 2022-08-28:
Ok, I agree for the positions. In fact there is two paradigms:

- Either we use the movement (like I set it). ie: you want to go "front" and thus increase position in Y+
- Either we use the position (like you propose it). ie: you want to go from the back.

At the end it's the same thing. But ok for me to change it with the new variables names. I'm starting a branch for this one and will do a PR to discuss on it :)

Frix-x, 2022-08-31:
Improvements were made in #15

closed 2022-08-31 by Frix-x