Frix-x/klippain #17: Improve README

Frix-x, 2022-08-28:
Now that the config is becoming more generic and used as is by multiple people (at least elpopo and me) on multiple machines (at least Voron V2.4 machines and also a TriZero machine), the REAME need to be improved to become more focused on global install instructions

d5aint, 2022-09-15:
Let me know what I can do to assist.
With the changes recently made, this is becoming an excellent config.

Frix-x, 2022-09-16:
Thank you for the proposal ! I was on holiday for a couple of days and will be back to it next week :)

So I'll rework the main README.md file to make it more generic and not focused only on "my machine". The goal is to describe the config tree, how to install it, what are the features, how to customize it and to adapt it to another machine. In the meantime, if you want to do a PR with a draft, feel free to do it :)

d5aint, 2022-09-17:
Thanks. will do that.

commit 60590077 (2022-11-20): global documentation update and moonraker path fix

fix #17
fix #27
fix #45
commit 7c181429 (2022-11-22): global documentation update, moonraker path fix, and new cfg files

fix #17
fix #27
fix #45
commit 7b0cd473 (2022-11-22): global documentation update, moonraker path fix, and new cfg files

fix #17
fix #27
fix #45
closed 2022-11-22 by Frix-x