Frix-x/klippain #40: Different Bed Size for "CALIBRATE" macro

ProphecyKt, 2022-10-30:
Hello,

At the moment, the macro "CALIBRATE" call "PA" and "Flow" calibration with a fixe bed size of 300x300.
```
{% if TYPE=="flow" %}
        [...]
        START_PRINT [...] SIZE=130_130_170_170
        [...]
    {% elif TYPE=="pressure_advance" %}
        [...]
        START_PRINT [...] SIZE=90_90_210_210
        [...]
```

I think it would be beneficial to include some simple parameters to call this macro with different bed sizes. (at least 300 and 350)

Thank

Frix-x, 2022-11-03:
Thanks for raising an issue. At the moment this file is more like an example of how to use it, not really a generic way to do it.
I will think about a way to do it automatically on every machine size and then modify this file accordingly :)

W141-ID, 2022-12-13:
souldn't it be possible to have a variable in the variables.cfg defining the bed size?

Frix-x, 2022-12-13:
No: bed size is already set in the stepper configuration. So better to just extract it from here :)

I just didn't had time to do it at the moment. But I'll do :)

Frix-x, 2023-04-16:
Hey, this is done in https://github.com/Frix-x/klippain/commit/490654b00d43b71fafb4cae23ba6e86c0fd97a1f
To be merged soon in the main branch :)

closed 2023-04-16 by Frix-x