Frix-x/klippain #45: Vibrations macros  does not works on newly installed Moonraker

fbeauKmi, 2022-11-18:
Hey,
As Moonraker changes files structures, the vibration macros does not work anymore on newly installed printer.

The shell command and plot_graphs.sh,  where pathes are hardcoded, irremediably yield to error. 

One way could be to convert all config to the new structure, another one to add symbolic links in the documentation. 

What do you think about it?

Frix-x, 2022-11-19:
Hello, thanks for raising an issue.

This is already in my to do list for the new config release. I plan to release it with a rework of this section to make it compatible with the new moonraker requirements (ie. change the paths and/or see how to be more flexible).

fbeauKmi, 2022-11-19:
Nice, I also identify an issue when user is not "pi". As it can be done writing Raspian Image, it could be a great improvement to tkae it in account.

Frix-x, 2022-11-19:
Yes this is already managed by the graph_vibrations.py script as you can add a `-k /home/xxxx/klipper_config` parameter to set the correct path for the script. It's the only place where the user name can be a problem. I agree though that it's not straightforward to use...

I definitely need to do something on this topic !

commit 60590077 (2022-11-20): global documentation update and moonraker path fix

fix #17
fix #27
fix #45
commit 7c181429 (2022-11-22): global documentation update, moonraker path fix, and new cfg files

fix #17
fix #27
fix #45
commit 7b0cd473 (2022-11-22): global documentation update, moonraker path fix, and new cfg files

fix #17
fix #27
fix #45
closed 2022-11-22 by Frix-x