Frix-x/klippain #64: Pin named z:.....

Fragmon, 2022-12-04:
In many .cfg files the pines are named z:[name of the alias].

with the z: the config dont work. I have to delete all z: then the config load.

yanfoo, 2022-12-10:
I also have an octopus as it is more and more common. The fix would be a few duplicated files (11 files I think) for the bigger boards.
otherwise we simply delete the z: as most people coming here to use this config may not be beginners, I think its a good baseline to work with and I do some minor tweaks to it. It is very clear an simple to find the one that has the z: aliase by looking in the SKR1.4_x2_V2.4.cfg file.

/config/heated_bed.cfg
/config/probe.cfg
/config/hardware/z/*.cfg
/config/hardware/filters/*.cfg
/config/hardware/fans/controller_fan.cfg
/config/hardware/accelerometers/adxl345_skr.cfg
/config/hardware/lights/fcob_white.cfg
/config/hardware/temperature_sensors/cabinet_temp.cfg

Frix-x, 2022-12-11:
Yes you're right. It's an easy fix.
However, I want to keep it like this at the moment as it's a lot easier for me to work on the config and test directly on my machine.

But as I said in my previous message, thanks to the sponsoring, I was able to afford an Octopus board that should come in a couple of days. So **I will remove all these `z:` as soon as my Octopus is installed** :)

closed 2023-03-29 by Frix-x