Frix-x/klippain #66: Relocating STATUS_LEDS to the initial Macro

W141-ID, 2022-12-05:
At the Moment it is placed at START_PRINT. Resulting in changing the light only when printing.

Moving the STATUS_LEDS call to the macro will result in changing lights everytime the macro is called.

Frix-x, 2022-12-06:
This is a good idea and I'll definitely do it :)

W141-ID, 2022-12-07:
> This is a good idea and I'll definitely do it :)

I have done a push including the needed changes. Feel free to comment on that.

Frix-x, 2022-12-16:
Hello,

Since I've merged #77 now. Can you do the LEDs PR again ? I still think it's a good addition to this config to make it more clean :)

Tell me if it's ok for you or I'll do it myself in the next couple of days

W141-ID, 2022-12-17:
Yes just waited for the merge. Will pr soon need to make some changes.

Frix-x, 2023-01-05:
Ok, I tested the PR #90 and everything is OK. I'll merge it and I think we can close this one :)

closed 2023-01-05 by Frix-x