Frix-x/klippain #73: Add calibration for Z belts tension

Frix-x, 2022-12-07:
This issue is to track and not forget to add a macro to calibrate and check the Z belts tension on a belted and multiple motor Z axis printer like the Voron 2.4

Planned work:
   - Add a macro that execute some move command to make the toolhead resonate in the Z direction (all the Z motors at the same time or if [force_move] is available by using it for 1x Z at a time if possible)
   - Add a python script to analyze the data for every measurement and graph something to be able to see the difference between all the Z belts

W141-ID, 2022-12-28:
Hey, since I've installed TAP on my printer I found this plugin https://github.com/sporkus/probe_accuracy_tests it doesn't work for me. But it might be the thing you are looking for

Frix-x, 2023-12-18:
Closing this one since this will now be managed by the Shake&Tune plugin and the preliminary tests doesn't give good results for now.

For everyone to know: Probe accuracy tests by Sporkus is compitble with Klippain since a couple of months and is working very well to validate the Z belts behavior :)

closed 2023-12-18 by Frix-x