Frix-x/klippain #74: Add calibration for the max flow

Frix-x, 2022-12-07:
This issue is to track and not forget about this.

Goal is to be able to quicly find out the max flow for every hotend and every material you have with only a click in Mainsail/Fluidd

fbeauKmi, 2022-12-10:
Hi,
I developped one inspired by your work. Feel free to use, adapt, comment as I'm not sure of the workflow you imagine.
[max_flow_calibrate](https://github.com/fbeauKmi/V2.3934/blob/main/klipper_config/macros/calibration/max_flow_calibrate.cfg). It was just a game to play with 3Dprint basic math. I only use it on ABS but on different hotend and nozzle size.
CU

Frix-x, 2022-12-11:
Hey thanks for this one ! It's exactly what I wanted to do :)

Can you open a PR ?

I also had in mind to maybe merge the current flow calibration macro with this idea and add a parameter to do this test or the normal calibration test. That way all the flow maths and shell printing movement will not be duplicated in two files. What do you think about it ?

@Fragmon if you want to try this beta macro, it should work the same way as the idea we were talking about on Discord :)

Fragmon, 2022-12-11:
@Frix-x @fbeauKmi 
Yes nice work. I open issues in the hub of Frédéric and try to convert that into the all in one cfg.

roykrikke, 2023-12-23:
Any progress on this item?