Frix-x/klippain #80: Update process

shlagwuk, 2022-12-10:
Hi,

Can you add an update process in your README.md ?
For instance, If I want to update to your last macros but keep my changes (printer.cfg, pinout etc), how should I do ?

Thanks!

yanfoo, 2022-12-10:
Being on Github you can clone this repo and work on your own. You can then compare his files when they are updated to your cloned repo this will not "commit" anything to his files but just on your own github files.

I used to do it this way with the github desktop app and honestly I find it faster to just edit the few files by hand.

Frix-x, 2022-12-11:
Yes, the magic of Git should allow an easy update while keeping your locally modified files. However, I agree that for Git beginners, it's not as straightforward.

So I'll add some documentation about it in the README :)

d5aint, 2023-01-08:
What you can do as well is open both files in notepad++ and compare the two.

shlagwuk, 2023-01-13:
> What you can do as well is open both files in notepad++ and compare the two.

Well yes, but that doesn't seem like the best way to use Git ;-)

I will manually upgrade to the last version and check my files one by one this time, and figure how to do it with git for the next commit!

closed 2023-03-29 by Frix-x