Frix-x/klippain #84: Lastest Klipper update breaks flow test

TravisWilder, 2022-12-12:
https://github.com/Klipper3d/klipper/commit/0c5c87d7c08c2b8aa2f248d81dd1003c5c7d5a7c

WIth the latest update Klipper will not work with the current Flow test and will stop on the G2/G3 commands

Frix-x, 2022-12-12:
Thanks for the hint. I'll correct that ASAP :)

fbeauKmi, 2022-12-14:
Don't be too prompt to fix that, there's probably a bug in this commit. When J or I = 0 its returns error missing coords. At least on my system :/

Frix-x, 2022-12-14:
This is discussed here : https://github.com/Klipper3d/klipper/pull/5937

fbeauKmi, 2022-12-15:
Fixed by this one https://github.com/Klipper3d/klipper/pull/5939 tested last night.  It's OK

Frix-x, 2023-01-05:
@fbeauKmi so I can close the issue ? Or is there some modifications to do on my macro ?
I still did not have time to update Klipper on my side but I'll do it next week if needed.

ShryuKt, 2023-01-05:
I think you can close, I tested your macro since the new klipper update and it's OK on my side.

fbeauKmi, 2023-01-05:
Hi, 
Nothing to change to your macros. Gcode_arcs behaves as it should since the last PR. You can close this one.

Frix-x, 2023-01-05:
Ok, thanks !
I'll close it now :)

closed 2023-01-05 by Frix-x