Frix-x/klippain #92: Add description in the Input Shaper Workflow doc for belt tension graphs

voidtrance, 2022-12-30:
There doesn't seem to be any documentation describing how to interpret the belt tension graphs that your scripts generate. It would be useful to know how to read the graphs:

* What do the peaks represent?
* What should the user be looking for?
* How to fix issues?

Thank you.

EricZimmerman, 2022-12-30:
It would be nice to have a read me with a bunch of compon graphs, what is good, what is bad, how to tweak etc. Same with input shaper really

Frix-x, 2022-12-31:
Hey,
Thanks for raising an issue on this topic. You're right, I get a lot of questions on the Discord about this subject and I will try to compile a documentation section to interpret them :)

Frix-x, 2023-03-19:
Hey, I'm currently working on some documentation. It's actually a WIP but you can have a look in #129

Frix-x, 2023-04-30:
Done in 2c41a777cc9a42c5ba481bbba610ba8490e7287c

Frix-x, 2023-06-11:
Closing as Klippain v4 is merged and released :)

closed 2023-06-11 by Frix-x