Frix-x/klippain #93: Relative Reference Index Calculation

JanB97, 2022-12-31:
The relative reference is calculated in the middle of the mesh.
If I understand correctly, relative index should be the point where you calibrated your nozzle, so for most people probably the middle of the bed. 
The calculation here assumes, that the nozzle was set up in the middle and your print area always is in the middle of the bed. 
Calculation should use coordinates as indicator and pick the probe point nearest to it.

To be clear, I tried calculating it myself but I cannot figure out a way to do it in Jinja, it's so limited.

Frix-x, 2023-01-06:
Hello,

Thanks for raising an issue about this. You're right: we want to get the RRI at the same position where the Z homing was done.

That's why my adaptive mesh macro can be called in two parts:
  - `COMPUTE_MESH_PARAMETERS` can just "compute" and define the mesh. This will also compute the RRI if needed (always in the center of the mesh)
  - Then you can use this `mesh_center` variable like I'm doing in any Z homing process. On my side I use the z automatic calibration plugin. But this can also work without it if you can do the Z homing on the bed like with a Voron TAP or BLTouch. So the homing can be done on the center point of the mesh, even if it's not on the middle of the bed.
  - Then `ADAPTIVE_BED_MESH` will effectively do the mesh as it was calculated at step one

JanB97, 2023-01-07:
Yes but it's not what I meant.
This is what I came up with to approximate the reference point for people using a seperate z endstop.:

`
 {% set steps_x = (area_max_x - area_min_x)/(meshPointX -1)  %}
        {% set steps_y = (area_max_y - area_min_y)/(meshPointY -1)  %}

        {% set mesh_vals = namespace(index=-1, reference=0) %}
        {% set mesh_temp_vals = namespace(x=9999, y=9999) %}

        {% for y in range(1,meshPointY+1) %}
            {% for x in range(1,meshPointX+1) %}

                {% set mesh_vals.index = mesh_vals.index+1  %} #Setting Index

                {% if (y % 2 == 0) %}                          #Calc X Value
                  {% set calcX = area_max_x-(steps_x*(x-1))  %} 
                {% else %}
                  {% set calcX = area_min_x+(steps_x*(x-1))  %}        
                {% endif %}

                {% set calcY = area_min_y+(steps_y*(y-1))  %} #Calc Y Value
                
                {% set diffX = (midX-calcX)*(1 if midX-calcX >= 0 else -1) %} #Calculate Differences
                {% set diffY = (midY-calcY)*(1 if midY-calcY >= 0 else -1) %}

                {% if (diffX <= mesh_temp_vals.x) and  (diffY <= mesh_temp_vals.y) %} #Setting Reference Value and Temp Data
                    {% set mesh_temp_vals.x = diffX %}
                    {% set mesh_temp_vals.y = diffY %}
                    {% set mesh_vals.reference = mesh_vals.index %}
                    #{action_respond_info("Ref Set!")} 
                {% endif %}

                #{action_respond_info("Diff: ------ X: %i | Y: %i" | format(diffX,diffY))} 
                #{action_respond_info("Index: %i --> X: %i | Y: %i" | format(mesh_vals.index, calcX, calcY))} 
            {% endfor %}
        {% endfor %}
        
        #{action_respond_info("Ref:     %i" | format(mesh_vals.reference))} 
        #{action_respond_info("Points:  %i" | format(meshPointY*meshPointX))} 
       ### Probe ###
       BED_MESH_CALIBRATE_BASE mesh_min={area_min_x},{area_min_y} mesh_max={area_max_x},{area_max_y} probe_count={meshPointX},{meshPointY} relative_reference_index={mesh_vals.reference}`

I'll close this, because I got it to work good enough for me. Thanks. :)

closed 2023-01-07 by JanB97