Frix-x/klippain #94: Voron TAP support

W141-ID, 2023-01-02:
I managed to change the code a bit to work with the voron tap but there are some changes needed to have a general approach.
Here are a few things I have found that need to be done:

- probe action codes need a if voron tap enabeled statement
- qgl has attach_prob & dock_brobe in them, needs to be more general for probes that don't need docking (it seems like the overrides are just for dockable probes? If so then add them to a dockable_probe.cfg instead of including it through the printer.cfg)
- heating up the probe to 150deg before probing needs to be added to the start_gcode or to the general gcode, maybe let the user decide?

Nothing really devestating just a few small additions/changes.

I can work on that if it's alright with you.

MarcHerzig, 2023-02-16:
are there some news about the voron TAP?

Frix-x, 2023-02-16:
Yes, everything is done regarding the probe management on the newer v3 version that I'm currently working on.
This v3 version will allow the use of TAP, but will also stay compatible with the classical "inductive probes" or "dockable probes".

You can have a look at the v3.0.0-alpha branch. But be aware that it's not ready for production yet as there is still other features and documentation that I need to add before the release :)

Frix-x, 2023-03-20:
Hello AMFkuna, thanks for your comments!

Regarding the TAP probing temperature, I think I could add a variable to force heating up to the safe temperature every time a probe is needed. So that way people could choose to let it like it is today or heat up everytime as you have done it.

Regarding the modification of the end_print to do a clean, be careful as this is quite dangerous if you are printing big parts. You could make a crash, that's why I'm not a fan of this...
But if some other user want to do this also, I could add it as a variable to enable a final clean in the print_end with lot of warnings.

Final note, in the meantime you can perfectly do what you want by using the overrides mechanism. Just copy/paste my version of the full macro in your overrides.cfg file and modify it as you need. This new macro will override and replace mine. Then when I update the repo you will be able to remove the overrides (or keep them if you prefer) :)

AMFkuna, 2023-03-20:
To clean the nozzle at the end of the print, my Stealthburner takes 37mm off the print surface on the Y axis.

AMFkuna, 2023-03-21:
Hi, the dangerous cleaning of the nozzle after the end of the print can be disabled using the SIZE variable for the adaptive bed mesh from the START_PRINT macro. What do you think?

closed 2023-03-29 by Frix-x