Frix-x/klippain #100: ERCF python script

d5aint, 2023-01-08:
Can you include the ercf.py script in the scripts folder?
Will need to include instructions as well for others to move it to the klipper/klippy/extra folder.

Frix-x, 2023-01-08:
This is a good idea.

And this make me think about making all in one install script that would install all the scripts for the IS workflow, vibrations, ERCF, moving all the files, doing the chmod, etc... It would allow an easier install procedure for everything... I need to think about it :)

d5aint, 2023-01-08:
this is off topic of this issue, but also the auto z-calibration script as well.

Frix-x, 2023-01-08:
I've changed the issue title to keep track of everything here. It will be easier to not forget that way 😅

sebadue, 2023-01-11:
Why should that be a good idea? Separately maintained scripts should be individually installed from their GitHub repositories and symlinked to the extras folder. You should be responsible for updating this repo if ercf or z_calib is updated.
Or did I get anything wrong here?

Frix-x, 2023-01-11:
Oh no no: I don't want to integrate the external scripts here (ERCF & auto Z calibration). I want to do an install script that fetch and download them directly from their respective repositories and do automatically the symlinks (or call their own install scripts).

sebadue, 2023-01-11:
Ah okay ;) Nevermind, I didn‘t get that.

d5aint, 2023-01-11:
Correct, would be too much work to constantly be monitoring them for updates.
Just need to make the end user aware that these scripts are required and how to get them.
But a git install would be perfect.

closed 2023-03-29 by Frix-x