Frix-x/klippain #98: Spider_v1.cfg - EXP1_1 should be PC9 instead of PE9

D1rect10ns, 2023-01-08:
https://github.com/Frix-x/klipper-voron-V2/blob/82bfb208768d80ab3b1546228d359edc7e79d1b4/config/mcus/spider_v1.cfg#L46

EXP1_1 is incorrectly defined as PE9 but should be PC9 ([output_pin beeper]), as PE9 is for X_ENABLE. In it's current state the spider_v1.cfg will throw the following error:

`pin X_ENABLE is an alias for EXP1_1`

Frix-x, 2023-01-08:
Hey, thanks for raising an issue here :)
I'm not using a spider board at the moment on my machines and there could be some mistakes. I'll investigate and correct the error !

@elpopo-eng I know you are using a spider board. What are you thoughts on that ?

Benoitone, 2023-01-08:
Base on this, I think there is an inversion EXP1/EXP2 too.
[image]

D1rect10ns, 2023-01-08:
I can confirm something isn't quite right with the EXP pins defined for the spider_v1.cfg as I get no display with the default config and a Fysetc mini12864. I do if I manually edit the mini12864.cfg file with the following:

```
[display]
lcd_type: uc1701
cs_pin: PC11
a0_pin: PD2
rst_pin: PC10
encoder_pins: ^PC7,^PC6
click_pin: ^!PA8
contrast: 63
spi_software_mosi_pin: PA7
spi_software_miso_pin: PA6
spi_software_sclk_pin: PA5

[output_pin beeper]
pin: PC9
```
The neopixel_mini12864.cfg file also needed amending:

```
[neopixel fysetc_mini12864]
pin: PC12
```
I'll work out what's going on and report back.

Frix-x, 2023-01-09:
Ok, thanks for looking at it.

It's indeed quite complicated for me to work on that based on the fact that I do not have any spider board neither any mini12864 on hand right now.
If you could give me some insights, I'll be happy to correct the files or even accept a PR from your side :)

elpopo-eng, 2023-01-09:
Hi @Frix-x Frix-x,
As @Benoitone shows EXP1.1 should be PC9 not PE9.
Be careful, the serigraphy of the spider 1.0 is reversed

D1rect10ns, 2023-01-09:
This is what worked for me in the spider_v1.cfg:

```
# EXP1 / EXP2 (display) pins
    # EXP1 header
    EXP1_1=<MISO>, EXP1_2=<SCK>,
    EXP1_3=PC6, EXP1_4=<SS>,
    EXP1_5=PC7, EXP1_6=<MOSI>,    # Slot in the socket on this side
    EXP1_7=PB10, EXP1_8=<RST>,
    EXP1_9=<GND>, EXP1_10=<5V>,

    # EXP2 header
    EXP2_1=PC9, EXP2_2=PA8,
    EXP2_3=PC11, EXP2_4=PD2,
    EXP2_5=PC10, EXP2_6=PC12,      # Slot in the socket on this side
    EXP2_7=PD0, EXP2_8=PD0,
    EXP2_9=<GND>, EXP2_10=<5V>,
```
Here's my mini12864.cfg contents, with comments on pins for the Fysetc mini12864. The BTT version should be the same:
```
[display]
lcd_type: uc1701
cs_pin: EXP2_3
a0_pin: EXP2_4
rst_pin: EXP2_5
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP2_2

##cs_pin: PC11
##a0_pin: PD2
##rst_pin: PC10
##encoder_pins: ^PC7,^PC6
##click_pin: ^!PA8
contrast: 63
## Some micro-controller boards may require an spi bus to be specified:
#spi_bus: spi
## Alternatively, some micro-controller boards may work with software spi:
#spi_software_miso_pin: EXP2_1
#spi_software_mosi_pin: EXP2_6
#spi_software_sclk_pin: EXP2_2
spi_software_mosi_pin: PA7
spi_software_miso_pin: PA6
spi_software_sclk_pin: PA5

[output_pin beeper]
pin: EXP2_1
#pin: PC9
```

Frix-x, 2023-01-11:
Thanks, I'll do the correction :)

commit b3fa7392 (2023-01-23): fixed the display configuration based on issue #98

close #98
commit d86aec2b (2023-03-29): fixed the display configuration based on issue #98

close #98