Frix-x/klippain #102: Update readme file to be in accordance with macro name

zebons, 2023-01-15:
In the readme file, we can find an exemple of the Gcode Start command for Superslicer :
START_PRINT EXTRUDER_TEMP={first_layer_temperature[initial_extruder] + extruder_temperature_offset[initial_extruder]} BED_TEMP=[first_layer_bed_temperature] MATERIAL=[filament_type] CHAMBER=[chamber_temperature] SIZE={first_layer_print_min[0]}_{first_layer_print_min[1]}_{first_layer_print_max[0]}_{first_layer_print_max[1]}

But the macros were renamed to PRINT_START and END_PRINT, so it'd be better to update the readme file to respect macro's names

d5aint, 2023-01-16:
Both of the macros that I see are [gcode_macro START_PRINT] and [gcode_macro END_PRINT].

zebons, 2023-01-16:
Sorry, you're right and the readme is already correct. I was testing the Bambulab Soft Fever slicer  and thinking that I had copy the entire command from the readme file but I only change a part of that.  So it's my fault and I'll close that unusefull issue.

closed 2023-01-16 by zebons