Frix-x/klippain #107: Error trying to plot graph

bpbastos, 2023-01-26:
I'm getting this error when trying to run BELTS_SHAPER_CALIBRATION:

`Traceback (most recent call last):
File "/home/pi/klipper/scripts/graph_accelerometer.py", line 259, in
main()
File "/home/pi/klipper/scripts/graph_accelerometer.py", line 218, in main
datas = [parse_log(fn, opts) for fn in args]
File "/home/pi/klipper/scripts/graph_accelerometer.py", line 218, in
datas = [parse_log(fn, opts) for fn in args]
File "/home/pi/klipper/scripts/graph_accelerometer.py", line 22, in parse_log
if not header.startswith('freq,psd_x,psd_y,psd_z,psd_xyz'):
UnboundLocalError: local variable 'header' referenced before assignment`

Frix-x, 2023-01-26:
Hey thanks for raising an issue here :)

First of all: are you `maciej_teledepresant V2.4821` on the Voron Discord ? If it's not the case, then there is at least to people with this bug.

This is an error thrown by the original klipper script (not mine) that means that there is no file to work on and analyze. So it could be because my script failed to move the files at the correct place.
Can you verify if there is some `xxxx_A.csv` and `xxxx_B.csv` files in the adxl_results/belts folder ?

bpbastos, 2023-01-26:
> Hey thanks for raising an issue here :)
>
> First of all: are you `maciej_teledepresant V2.4821` on the Voron Discord ? If it's not the case, then there is at least to people with this bug.
>
> This is an [...]

Hello Frix-x, thank you for this awesome macro!

 I'm not `maciej_teledepresant V2.4821`, I'm running it from a Two Trees Sapphire Plus with klipper v0.11.0-86-g6026a99a.

Yes, there is the following files in /config/adxl_results/belts:

[image]

Frix-x, 2023-01-30:
Hey, I've made some reasearch regarding this and unfortunately it looks like a bug from Klipper code. Here is a PR that should solve the issue : https://github.com/Klipper3d/klipper/pull/6003

What I find very strange is that this code from Klipper seems to have no change since december 2021. So this bug should be present since this date but it was working fine on my side (and at a lot of other users also) with a much more recent klipper version... So not sure it will solve your issue but you can try to add the indentation on those two lines and see if it helps... ?!

Please keep me informed :)

Zannys, 2023-02-24:
I tried use belt shaper function but I receive this error:

22:09
Command {plot_graph} finished
22:09
Graphs created. You will find the results in /home/fly/printer_data/config/adxl_results
22:09
File "/home/fly/klipper/scripts/graph_accelerometer.py", line 22, in parse_log
if not header.startswith('freq,psd_x,psd_y,psd_z,psd_xyz'):
UnboundLocalError: local variable 'header' referenced before assignment
22:09
datas = [parse_log(fn, opts) for fn in args]
File "/home/fly/klipper/scripts/graph_accelerometer.py", line 218, in
datas = [parse_log(fn, opts) for fn in args]
22:09
main()
File "/home/fly/klipper/scripts/graph_accelerometer.py", line 218, in main
22:09
Traceback (most recent call last):
File "/home/fly/klipper/scripts/graph_accelerometer.py", line 259, in
22:09
Running Command {plot_graph}...:
22:09
echo: Graphs generation... Please wait a minute or two and look in the configured folder.
22:09
Writing raw accelerometer data to /tmp/raw_data_axis=1.000,-1.000_a.csv file

and when a look in folder on file have 0.1kb

[image]

Frix-x, 2023-02-27:
Hey thanks for the info!

So here is what I found: the Klipper code is good and there is no bug in it. The only problem is that the Klipper script can fail when the file is empty. And this looks to be the case for you as one of the file is 0.1kB.

I was not able to reproduce it on my side but I think it's related to the `mv` command being done a little bit too early when Klipper is also still writing in the file on his side. This could happens when using a bad/slow SD card.

In order to avoid problem, I currently do 2 seconds dwell time in my script to allow this kind of delay but maybe it's not enough for you. Try increasing it to somethink like 5-10 seconds and see if it helps ?

Here are the lines : 
https://github.com/Frix-x/klipper-voron-V2/blob/a3b33ae21b2f4c47bb86b0bc7b85f0e90eb47813/scripts/plot_graphs.sh#L58
https://github.com/Frix-x/klipper-voron-V2/blob/a3b33ae21b2f4c47bb86b0bc7b85f0e90eb47813/scripts/plot_graphs.sh#L73
https://github.com/Frix-x/klipper-voron-V2/blob/a3b33ae21b2f4c47bb86b0bc7b85f0e90eb47813/scripts/plot_graphs.sh#L87

Frix-x, 2023-05-01:
Hello, I wanted to try another idea without the use of lsof (from PR #128 ). Can you test the last version of the script that is pushed in https://github.com/Frix-x/klippain/commit/f1cf77a1f3381e749403b3c0a9b279db2d5db402 ?

FYI, I just added a sync && sleep 5 before the use of any function. I hope it will be sufficient to solve the problem :)

Esoterical, 2023-05-02:
Not sure if this is the same problem, but I just tried installing/running the input shaper graph stuff today. Running the belt shaper macro does run, and generates ~20MB csv files for each motor, but throws errors when trying to actually generate the graph. Not sure if It's an old/different numpy thing or what. A standard klipper "Test resonances" method to create graphs still works fine.
[image]

Frix-x, 2023-05-02:
> Not sure if this is the same problem, but I just tried installing/running the input shaper graph stuff today. Running the belt shaper macro does run, and generates ~20MB csv files for each motor, but throws errors when trying to actually generate the graph.

Yes, this is indeed the same issue. So my fix doesn't work correctly and I will need to work on something else. It's due to Klipper being a little bit too slow to release the file marker and start the script to early. Then my script try to move a file that is not already closed on Klipper side...

Frix-x, 2023-05-02:
So I tried another fix that simulate the behavior of lsof but without lsof :)
@Esoterical, can you try it: https://github.com/Frix-x/klippain/commit/e40edf2a26167bbbac07652c28c9ca312c61c51e ?

Esoterical, 2023-05-02:
> @Esoterical, can you try it: [e40edf2](https://github.com/Frix-x/klippain/commit/e40edf2a26167bbbac07652c28c9ca312c61c51e) ?

Huzzah! that worked for me, plotted the graph as expected. Thanks for the fix.
[image]

Frix-x, 2023-06-11:
Closing as Klippain v4 is merged and released :)

closed 2023-06-11 by Frix-x