Frix-x/klippain #110: [FR] a way to disable the clicky probe

RustedAperture, 2023-01-29:
Currently when running the probe_calibrate it will try to do a probe dock and mount. My solution recently was to just remove the code inside probe_base.cfg as I use a 8mm probe.

for generalizing the config I beleive it would be helpfult o simply disable the klicky probe docking.

Frix-x, 2023-01-29:
Hey, thanks for raising an issue here :)

I'm not sure to understand. What kind of probe are you running ?

FYI, the v2 version of this config is only compatible with the dockable probes like Klicky, Euclid, Quickdraw, etc...
But I'm currently working on a v3 version that will be more permissive with the additional support of Voron TAP and stock inductive probes.

RustedAperture, 2023-01-30:
Ok thank you for this info, I'm not using a dock able probe as mine is just a super pinda from a prusa. It is good to hear that version 3 will support more probes

closed 2023-01-30 by RustedAperture