Frix-x/klippain #111: Virtual Endstop support

fsaad1984, 2023-02-02:
It would be nice to have an option in the upcoming V3 to enable Virtual Endstops for X and Y. At the moment I hard coded it by setting the endstop pins to tmc2209_stepper_x:virtual_endstop and tmc2209_stepper_y:virtual_endstop.

Frix-x, 2023-02-02:
Thanks for the suggestion !
I'll have a look at how to do that properly :)

fsaad1984, 2023-02-02:
You are welcome, I am looking forward to new release!

Frix-x, 2023-02-27:
This was done in https://github.com/Frix-x/klipper-voron-V2/pull/104/commits/a9cdb861b42094e617404d088f33584d676527da :)

I'll keep this issue open until the v3.0.0-alpha branch is merged to follow everything

AMFkuna, 2023-03-20:
For correct sensorless  homing I add this to homing_override.cfg
`

    {% set RUN_CURRENT_X = printer.configfile.settings['tmc2209 stepper_x'].run_current|float %}
    {% set RUN_CURRENT_Y = printer.configfile.settings['tmc2209 stepper_y'].run_current|float %}
    {% set HOME_CURRENT = 0.7 %}

    {% if homing_first == "X" %}
        {% if X %} # Home x
            {% if verbose %}
                { action_respond_info("Homing X") }
            {% endif %}
            SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CURRENT}
            SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CURRENT}
            G28 X0
            G1 X{x_position_endstop + x_homing_backoff} F{homing_travel_speed}
            G4 P1000
            SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CURRENT_X}
            SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CURRENT_Y}
        {% endif %}
        {% if Y %} # Home y
            {% if verbose %}
                { action_respond_info("Homing Y") }
            {% endif %}
            SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CURRENT}
            SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CURRENT}
            G28 Y0
            G1 Y{y_position_endstop + y_homing_backoff} F{homing_travel_speed}
            G4 P1000
            SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CURRENT_X}
            SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CURRENT_Y}
        {% endif %}
    
    {% elif homing_first == "Y" %}
        {% if Y %} # Home y
            {% if verbose %}
                { action_respond_info("Homing Y") }
            {% endif %}
            SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CURRENT}
            SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CURRENT}
            G28 Y0
            G1 Y{y_position_endstop + y_homing_backoff} F{homing_travel_speed}
            G4 P1000
            SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CURRENT_X}
            SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CURRENT_Y}
        {% endif %}
        {% if X %} # Home x
            {% if verbose %}
                { action_respond_info("Homing X") }
            {% endif %}
            SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CURRENT}
            SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CURRENT}
            G28 X0
            G1 X{x_position_endstop + x_homing_backoff} F{homing_travel_speed}
            G4 P1000
            SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CURRENT_X}
            SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CURRENT_Y}
        {% endif %}
 `

Frix-x, 2023-03-20:
Thanks for the comments.

Yes I'm already aware of that and plan to add a variable "current_factor" to manage this. The goal is to set it to whatever percent is needed (like 60%) and this will set the current to 60% of the nominal current during homing.

But as a side note, this is something that I currently **do not use** in my case as I found out that setting correctly the `driver_SGTHRS` value is enough to get a proper homing without destroying the axis limits.

AMFkuna, 2023-03-20:
I am using the MOONS motors from the Formbot kit on my V2.4.
Current is set to 0.8A and SGTHRS to 110 on both axes.
When homing the axes separately, there is no problem, but if I do G28 X Y, or just G28, the second axis in the sequence, according to the variable in variables.cfg, always goes in the opposite direction. After adding homing current to homing_override the problem did not occur. Your code also lacked the G4 P1000 pause for proper sensorless homing.
BTW. Thanks for your great work!

Frix-x, 2023-03-25:
Thanks for the feedback. I've made a PR to address your needs. Can you check and review it before merging ?

https://github.com/Frix-x/klipper-voron-V2/pull/130

closed 2023-03-29 by Frix-x