Frix-x/klippain #112: Mellow Fly-sb2040 CANboard support

fsaad1984, 2023-02-02:
This CANboard is pretty popular and used by many in the community. It would be nice to have this mcu added to the config/mcu folder.

Frix-x, 2023-02-02:
Thanks,

I'll add it as well. If you want to do a PR about that, I'll be happy to accept it :)

wtsexton, 2023-02-05:
Here is the configuration I use for one I recently installed.

```
[mcu toolhead]
##--------------------------------------------------------------------
canbus_uuid:  b245dace1fa4
##--------------------------------------------------------------------

[board_pins mcu_sb2040]
mcu: toolhead
aliases:
# steppers
   E_STEP=gpio9,  E_DIR=gpio10,  E_ENABLE=gpio7,  E_TMCUART=gpio8,
   X_STOP=gpio29,
# hotend
   E_HEATER=gpio6, E_TEMPERATURE=gpio27, E_FAN=gpio14,
   PART_FAN=gpio13,
# probe
   PROBE_INPUT=gpio28,
# light
   STATUS_NEOPIXEL=gpio12,
# adxl
   ADXL_CS=gpio1, ADXL_SCLK=gpio0, ADXL_MOSI=gpio3, ADXL_MISO=gpio2

```

Frix-x, 2023-02-06:
Thanks for the pinout :)

fsaad1984, 2023-02-06:
Hi @flatbrotato, thanks for your comment. This config is indeed similar to what I use. I also defined the chamber temp sensor although this gives an inaccurate reading due to the sensor's placement:

# chamber
    CHAMBER_TEMPERATURE=gpio26

I am not super confident in how to push a PR, so feel free to do so if you want.

flatbrotato, 2023-02-06:
> Hi @flatbrotato, thanks for your comment. This config is indeed similar to what I use. I also defined the chamber temp sensor although this gives an inaccurate reading due to the sensor's [...]

I did the same, however I defined it else where so I can see it.

```
[temperature_sensor Toolhead_Chamber]
sensor_type = ATC Semitec 104GT-2
sensor_pin = toolhead:gpio26
```

Frix-x, 2023-02-26:
This is done in commit https://github.com/Frix-x/klipper-voron-V2/pull/104/commits/058469ef3ad29324aa1d2716f1e0b9d281e8a4ce :)

closed 2023-03-29 by Frix-x