Frix-x/klippain #124: Vibrations measurements command not found

EduardoMDSousa, 2023-02-27:
Good morning, I have this problem, I can't understand why it always gives this error.

read write awards are ok i can edit the plot_graphs.sh on fluid and save 

I'm using a mks pi , the directory is ~/mks instead of ~/pi

effectively it seems that it finds the file but as I am a nuvato on linux I am not understanding if you can help me thank you

// Resonances data written to /tmp/resonances_y_20230227_152504.csv file
23:27:26  // Running Command {plot_graph}...:
23:27:26  // /home/mks/printer_data/config/scripts/plot_graphs.sh: line 7: $'\r': command not found
23:27:26  // /home/mks/printer_data/config/scripts/plot_graphs.sh: line 14: $'\r': command not found
23:27:26  // /home/mks/printer_data/config/scripts/plot_graphs.sh: line 24: $'\r': command not found
23:27:26  // /home/mks/printer_data/config/scripts/plot_graphs.sh: line 31: $'\r': command not found
23:27:27  // /home/mks/printer_data/config/scripts/plot_graphs.sh: line 32: $'\r': command not found
23:27:27  // /home/mks/printer_data/config/scripts/plot_graphs.sh: line 39: $'\r': command not found
23:27:27  // /home/mks/printer_data/config/scripts/plot_graphs.sh: line 40: $'\r': command not found
23:27:27  // /home/mks/printer_data/config/scripts/plot_graphs.sh: line 44: $'\r': command not found
23:27:27  // /home/mks/printer_data/config/scripts/plot_graphs.sh: line 45: syntax error near unexpected token `$'{\r''
// /home/mks/printer_data/config/scripts/plot_graphs.sh: line 45: `function plot_shaper_graph {
'
23:27:27  // Command {plot_graph} finished

#################################################################################################################
RESULTS_FOLDER=~/mks/printer_data/config/adxl_results # Path to the folder where storing the results files
SCRIPTS_FOLDER=~/mks/printer_data/config/scripts # Path to the folder where the graph_vibrations.py is located
KLIPPER_FOLDER=~/mks/klipper # Path of the klipper main folder
STORE_RESULTS=0 # Number of results to keep (older files are automatically cleaned). 0 to keep them indefinitely
#################################################################################################################

Frix-x, 2023-02-28:
Hello,

This is not really a bug but a mistake from your editor. `'\r': command not found` means that the end of line are Microsoft windows style and not Unix style. So you need to correct the file to remove all those `\r`

Not sure it will works, but you can try in the Fluidd editor on the plot_graph.sh file to remplace them. Please hit `Ctrl + F` and configure it like that (do not forget the `.*` button). And remove all occurence of `\r` if you find some of them (not sure the editor allow to see them but it's worth a try)
[image]

EduardoMDSousa, 2023-02-28:
Thank you very mutch for Replay, i Will try that thank you

EduardoMDSousa, 2023-03-02:
Hi, I finally managed to get scrypt working but now I have this error

 23:29:14  // Writing raw accelerometer data to /tmp/raw_data_axis=1.000,-1.000_a.csv file
23:29:14  echo: Graphs generation... Please wait a minute or two and look in the configured folder.
23:29:14  // Running Command {plot_graph}...:
23:29:20  // Traceback (most recent call last):
// File "/home/mks/klipper/scripts/graph_accelerometer.py", line 259, in
// main()
// File "/home/mks/klipper/scripts/graph_accelerometer.py", line 218, in main
// datas = [parse_log(fn, opts) for fn in args]
// File "/home/mks/klipper/scripts/graph_accelerometer.py", line 218, in
// datas = [parse_log(fn, opts) for fn in args]
// File "/home/mks/klipper/scripts/graph_accelerometer.py", line 22, in parse_log
// if not header.startswith('freq,psd_x,psd_y,psd_z,psd_xyz'):
// UnboundLocalError: local variable 'header' referenced before assignment
23:29:20  // Graphs created. You will find the results in /home/mks/printer_data/config/adxl_results
23:29:20  // Command {plot_graph} finished

the csv files are in the belts folder but it didn't create the images I believe due to the error, can you tell me what could be wrong?
I don't use raspberry, I use mks pi.
I can calibrate the input shaper I usually use the automatic mode SHAPER_CALIBRATE and it works,
I actually tried to convert the csv by the normal method and I also got this error

mks@mkspi:~$ ~/klipper/scripts/graph_accelerometer.py /home/mks/printer_data/config/adxl_results/belts/belt_20230301_152913_A.csv
Traceback (most recent call last):
  File "/home/mks/klipper/scripts/graph_accelerometer.py", line 259, in <module>
    main()
  File "/home/mks/klipper/scripts/graph_accelerometer.py", line 218, in main
    datas = [parse_log(fn, opts) for fn in args]
  File "/home/mks/klipper/scripts/graph_accelerometer.py", line 218, in <listcomp>
    datas = [parse_log(fn, opts) for fn in args]
  File "/home/mks/klipper/scripts/graph_accelerometer.py", line 22, in parse_log
    if not header.startswith('freq,psd_x,psd_y,psd_z,psd_xyz'):
UnboundLocalError: local variable 'header' referenced before assignment

thank you

Frix-x, 2023-03-02:
Ok, now this look like a duplicate of #107 
Please see in this issue if my last message can solve your problem. If it's the case, I'll increase this dwell time :)

EduardoMDSousa, 2023-03-03:
[image]

I tried what it says in #107, it really looks like the same error since I always have A at 0 kb.
but
I already put 120 s, the error has changed

 20:23:57  // Re-enabled [input_shaper]
20:23:59  // Writing raw accelerometer data to /tmp/raw_data_axis=1.000,-1.000_a.csv file
20:23:59  echo: Graphs generation... Please wait a minute or two and look in the configured folder.
20:23:59  // Running Command {plot_graph}...:
20:26:03  // Traceback (most recent call last):
// File "/home/mks/klipper/scripts/graph_accelerometer.py", line 277, in
// main()
// File "/home/mks/klipper/scripts/graph_accelerometer.py", line 238, in main
// datas = [parse_log(fn, opts) for fn in args]
// File "/home/mks/klipper/scripts/graph_accelerometer.py", line 238, in
// datas = [parse_log(fn, opts) for fn in args]
// File "/home/mks/klipper/scripts/graph_accelerometer.py", line 28, in parse_log
// data = np.loadtxt(logname, skiprows=1, comments='#', delimiter=',')
// File "/usr/lib/python3/dist-packages/numpy/lib/npyio.py", line 1079, in loadtxt
// next(fh)
// StopIteration
20:26:03  // Graphs created. You will find the results in /home/mks/printer_data/config/adxl_results
20:26:03  // Command {plot_graph} finished

EduardoMDSousa, 2023-03-06:
Update Axis_shape_calibration is working now with 240s dwell time

i have 1 question, you srypt use my imput shape configs, setings , or in the scrypt is there a max_smoothing: 0.05 parameter , becase if i do SHAPER_CALIBRATE (auto mode) i have big diferrence on results like 2200 with your macros and 13800 with SHAPER_CALIBRATE

thank you

Frix-x, 2023-03-07:
Wow 240s !!! What brand/model of SD card do you have in your Pi ? And what model of Pi is it ? This look like a very slow SD card !

Regarding your question, my macro and scripts are basically just doing :

```
# In the Gcode macro:
TEST_RESONANCES AXIS=X

# Then in the bash script:
~/klipper/scripts/calibrate_shaper.py measurement_file.csv -o result_graph_file.png
```

So I'm not calling the SHAPER_CALIBRATE "auto-way" as I would have no control on the files. It´s like someone would do using the standard manual way, but automated with some file management added to make it easy for everyone.

EduardoMDSousa, 2023-03-08:
I use mks-pi with emc 8gb , 

The question on command is becase i think the smooth time is bigger than the One i have in the printer.cfg is 0.05 

I AM thinking the auto shaper is reading the variable sm time on printer.cfg but other command s dont , but its only my understanding I dont know  , it was wy i hask . 

I still cant do belts , it do belt b but belt A gets 0kb 😥, i can plot the CSV with the command but ONLY the graf came no aditional info , on Axis commando to , i Will re install all on my pi2 it was working fine to test the scripts

Frix-x, 2023-04-30:
Closing this in favor of #107

closed 2023-04-30 by Frix-x