Frix-x/klippain #131: ValueError

spikeygg, 2023-03-26:
@Frix-x, hey, I just tried to run your `VIBRATIONS_CALIBRATION` routine and I ran into a Python `ValueError`. I think it's because of how you're RegEx'ing with this line: https://github.com/Frix-x/klipper-voron-V2/blob/a3b33ae21b2f4c47bb86b0bc7b85f0e90eb47813/scripts/graph_vibrations.py#L176

turns out my login name (spikeygg) matches the start of your regex and the 'n' fits for `resonance` so instead of getting the speed for the `speed` variable, I get: `ikeygg/reso` and eventually: `// ValueError: invalid literal for int() with base 10: 'ikeygg/reso'`

This is probably because RegEx is greedy by default grabbing the first thing it finds. If I were doing this I'd breakout the basename with os first, that way you can be sure your operating on the filename and not the whole path. Like this:

`speed = re.search('sp(.+?)n', os.path.basename(logname)).group(1)`

btw: I made this change locally on my printer and ran it through the script and it worked.

Frix-x, 2023-03-27:
Hello,

Thank you for raising this issue! This is indeed something I did not think about when writing this piece of code.

Do you want to open a PR ?

Thanks!

spikeygg, 2023-03-27:
> Do you want to open a PR ?

Sure! I tried but I got this:
`ERROR: Permission to Frix-x/klipper-voron-V2.git denied to spikeygg.`

Frix-x, 2023-03-27:
> > Do you want to open a PR ?
> 
> Sure! I tried but I got this: `ERROR: Permission to Frix-x/klipper-voron-V2.git denied to spikeygg.`

Oh yes. I was thinking you were familiar with git. Sorry.
So in order to do a PR, you will need to fork the repository on your account. Then do the modification on your own fork (were you have all the rights to do what you want). And only then you will be able to open a PR from the Github interface to ask to merge your fork into my repository

spikeygg, 2023-03-29:
Let me give that a try.

Frix-x, 2023-04-11:
This was done in the develop branch (to be merged soon) in commit: https://github.com/Frix-x/klippain/commit/2143d4585b33aa1eadc8d968dac38cc71371a8e1

closed 2023-04-16 by Frix-x