Frix-x/klippain #134: BTT_Octopus definition pins

yanfoo, 2023-04-02:
Hello, I notice the user_templates of the BTT_Octopus uses E_HEATER=MCU_HE0 & BED_HEATER=MCU_HE1 being MCU_HE0=PA2 & MCU_HE1=PA3 respectively, in most case the user should/would have used MCU_BED0=PA1 for BED and not PA3.
I use a Rapido and wired the heater to PA1 so when I simply swapped over the MCU_HE0 & MCU_HE1 in my MCU.cfg thats when I noticed it.

Furthermore, should I copy my entire MCU.cfg content or just my swapped terminals (lights, nevermore, neopixels) and serial path in overrides.cfg or the MCU.cfg also doesnt get errased on updates?

I also use a toolhead thermistor (I never really confirmed how accurate it was but yeah still using it), it seems for this LDO toolhead board I have not manage to add it in my overrides.cfg successfully  it will say and I have to modify /config/hardware/temperature_sensors/chamber_temp.cfg and use the  following making the Klippain dirty obviously.

[gcode_macro _USER_VARIABLES]
variable_chamber_temperature_sensor_enabled: True
variable_chamber_temperature_sensor_name: "Chamber"
gcode:

[thermistor CMFB103F3950FANT]
temperature1: 0.0
resistance1: 32116.0
temperature2: 40.0
resistance2: 5309.0
temperature3: 80.0
resistance3: 1228.0

[temperature_sensor Chamber]
sensor_type: CMFB103F3950FANT
sensor_pin: CHAMBER_TEMPERATURE
min_temp: 0
max_temp: 100
gcode_id: chamber_th

Frix-x, 2023-04-03:
Hey thanks for raising an issue here :)

So to answer your questions:

> Hello, I notice the user_templates of the BTT_Octopus uses E_HEATER=MCU_HE0 & BED_HEATER=MCU_HE1 being MCU_HE0=PA2 & MCU_HE1=PA3 respectively, in most case the user should/would have used [...]

This is normal as I tried to be as close as possible to the official Voron documentation as this config is mainly done for Voron printers (even if it's works for other printers also). So that's why it's like that but as you have seen, it's quite easy to update it. BTW, you can do it in the mcu.cfg file as this one is yours and will not be erased (as all the other files in green on this picture).

[image]

> I also use a toolhead thermistor (I never really confirmed how accurate it was but yeah still using it), it seems for this LDO toolhead board I have not manage to add it in my overrides.cfg [...]

I don't understand why it's is not working... But I think you can enable in the printer.cfg the `chamber_temp.cfg` file and then only add to your overrides.cfg:

```
[thermistor CMFB103F3950FANT]
temperature1: 0.0
resistance1: 32116.0
temperature2: 40.0
resistance2: 5309.0
temperature3: 80.0
resistance3: 1228.0

[temperature_sensor Chamber]
sensor_type: CMFB103F3950FANT
sensor_pin: CHAMBER_TEMPERATURE
```

Frix-x, 2023-04-10:
Stale issue, I'll close it for now. If you feel the need to reopen it, please do :)

closed 2023-04-10 by Frix-x