Frix-x/klippain #135: save_config

N3oBahamut, 2023-04-06:
why am I having issues saving?  even before you changed things I had problems.. have to  move my extruder and bed heater to printer.cfg for it to save but its not working when i do this either.  im saya that its conflicting.

Frix-x, 2023-04-06:
Hello,

This is indeed a known Klipper bug: if a config is split with some `[include xxx.cfg]` then SAVE_CONFIG cannot work anymore.
I have no solution at the moment and think that splitting the config is much more valuable than this SAVE_CONFIG system. I suggest you to open an issue on the Klipper Discourse: maybe if more people raise it, it will get fixed?

In the meantime, today when I do a PID or Z offset adjustment, I just manually copy the value from the Mainsail/Fluidd console into my overrides.cfg.

Frix-x, 2023-04-06:
I'll also add some notes in the documentation :)

N3oBahamut, 2023-04-07:
what do you do about the bed mesh?  the same?

Frix-x, 2023-04-07:
> what do you do about the bed mesh? the same?

Yeah in fact I think it's a bad habit to save mesh as they change a lot. For example if you use a flex plate and don't place it everytime at the same position, the mesh will be different each time.
So that's why in my Klippain config, I do a mesh in every printstart, everytime. But to avoid loosing time, I also do an adaptive [bed mesh](https://github.com/Frix-x/klippain/blob/main/docs/features/adaptive_bed_mesh.md).

Cisien, 2023-04-10:
Either klipper fixed this in a recent update, or i an doing something different that enables it. 

Assuming it's not a fix on klipper's side -- my theory is that you'll have to seed printer.cfg with a result of a SAVE_CONFIG that would otherwise be there. It would be great to understand if it's a klipper fix i'm seeing or something else. I'll try to validate later.

Frix-x, 2023-04-10:
> Either klipper fixed this in a recent update, or i an doing something different that enables it.
>
> Assuming it's not a fix on klipper's side -- my theory is that you'll have to seed printer.cfg with [...]

You're right. As already discussed on Discord (but copying here for the record): having a "seed" in the `printer.cfg` file is indeed allowing the system to work. But it's a fix that I will not be able to implement in Klippain as the seed would need to be customized depending on the user hardware (virtual probe offset vs physical probe and endstop position, bed mesh vs no bed mesh, etc...)

N3oBahamut, 2023-04-10:
can you explain what you mean by having a seed?  I'm still learning klipper and appreciate your work and everyone's input.

Cisien, 2023-04-10:
specifically, i'm referring to this specific text at the bottom of printer.cfg
```
#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#```

Cisien, 2023-04-10:
I have been able to confirm that SAVE_CONFIG is functional with only that text I placed above at the bottom of the printer.cfg using the latest (as of a few days ago) klipper release

Cisien, 2023-04-10:
Just for completeness sake, I did try it without any of the automatically generated SAVE_CONFIG text present, and it also worked.

The specific config I was testing was the result of Beacon's BEACON_CALIBRATE

Frix-x, 2023-04-11:
Wow, this is very good to know! Thanks for the testings :)
Do you want to do a PR to add it to the bottom of `printer.cfg`?

Frix-x, 2023-04-11:
Ok, this is fixed on the develop branch by this commit (will be merged to main soon): https://github.com/Frix-x/klippain/commit/b06c375b868042badf0b6934b67d32d35d0ddc7b

Benoitone, 2023-04-12:
@Cisien 
Which version of klipper?

For you, which variable does it work for saved by SAVE_CONFIG?

Benoitone, 2023-04-12:
@Cisien 
is it possible for you to post the content of your printer.cfg you use actualy?

For the default config of klippain, SAVE_CONFIG doesn't work for the z_offset I test it with Euclid probe and TAP probe with the last release of klipper (v0.11.0-173)

[image]

Cisien, 2023-04-12:
the printer.cfg on its own is bog standard

```
<a whole lot of comments and [include] sections omitted for brevity, you can compare it to many of the other printer.cfg's from klippain>

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [beacon model default]
#*# model_coef = 1.5557885534226927,
#*# 	  1.8778008527124102,
#*# 	  0.7567165232089866,
#*# 	  0.3440585301249553,
#*# 	  0.23876197429063728,
#*# 	  0.20266821408310468,
#*# 	  -0.0542998702249052,
#*# 	  -0.09878133588054372,
#*# 	  0.10399603926506155,
#*# 	  0.07711427606976333
#*# model_domain = 3.237936228000499e-07,3.336791225978232e-07
#*# model_range = 0.200000,5.000000
#*# model_temp = 25.475948
#*# model_offset = -0.53000
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	0.038496, 0.036716, 0.036488, 0.033203, 0.030080
#*# 	0.032755, 0.028819, 0.024031, 0.024290, 0.016309
#*# 	0.007133, 0.005486, 0.000000, 0.004096, -0.002058
#*# 	-0.011066, -0.009995, -0.012625, -0.013599, -0.017896
#*# 	-0.014728, -0.011678, -0.015980, -0.016643, -0.020765
#*# x_count = 5
#*# y_count = 5
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = bicubic
#*# tension = 0.2
#*# min_x = 144.0
#*# max_x = 195.0
#*# min_y = 146.0
#*# max_y = 198.0
```

try adding these lines to the bottom of your printer.cfg

```
#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*#[probe] and comment out your existing probe's `z_offset`
```

Benoitone, 2023-04-13:
Hello,

**1.** So you use Beacon probe and don't use the include probe from klippain in your printer.cfg?

```
##### Probe configuration --------------------------------------------
# [include config/hardware/probes/inductive_probe.cfg]
# [include config/hardware/probes/dockable_probe.cfg]
# [include config/hardware/probes/tap_probe.cfg]
# --------------------------------------------------------------------
```

**2.** and where did you insert the default config for your beacon? In overrides.cfg or in printer.cfg?

**3.** and you didn't use SAVE_CONFIG command from Klipper but BEACON_CALIBRATE from the external module beacon.py?

I think your fix is ok for YOU but it's a special case et and does not work for most klippain users.

**Can you try to experience the "bug" of klipper by calibrating the pids (extruder or heater_bed)???**  https://docs.vorondesign.com/build/startup/#pid-tune-bed--hotend

Cisien, 2023-04-13:
Here is my beacon config: https://github.com/Cisien/klippain/blob/main/config/hardware/probes/beacon.cfg

it is included just like any other probe config

Benoitone, 2023-04-13:
to make it work you must CUT (REMOVE or COMMENT]:
```
[extruder]
control: pid
pid_kp: 26.015
pid_ki: 1.770
pid_kd: 95.603

[heater_bed]
control: pid
pid_kp: 38.771
pid_ki: 1.746
pid_kd: 215.177
```
from overrides.cfg
and PASTE (ADD) them to your printer.cfg

Cisien, 2023-04-13:
Mystery solved, you can stop being condescending now.

Benoitone, 2023-04-13:
> Mystery solved, you can stop being condescending now.

so you're ok, your fix didn't work for official klippain config?

Frix-x, 2023-04-15:
Hey, hello!

So here is a post to sum up with some testings on my side:
  1. As of now, `SAVE_CONFIG` doesn't work natively with Klippain. It's a known thing of Klipper when the section to be saved is in a file included by `[include xxx.cfg]`
  2. Putting the element you want to save in the `printer.cfg` like PIDs, make it work for this element only. If PIDs are defined in one of the included file, then we fall in case 1 and it fail. But this is not something conceivable in Klippain as there is some configuration included or not depending on every user (mainly on the Z offset settings).
  3. `SAVE_CONFIG` will work for elements that doesn't need to be set (like a mesh) as the saved values will not need to "override" something in one of the includes (ie. we do not set a mesh points/values in our config)
  4. Adding the seed helps to SAVE things without a bug. But if for some reason you have a let's say the PIDs set in the overrides.cfg file (as I'm doing currently), you will have a problem as the saved values will not be used (this could be confusing).
  5. Last point, when using a new parameter from another thing like your beacon calibrate, it's like on point 3 if the element doesn't have a proper initialisation of its config. Or fall in the point 4 and need the seed to work.

So, with these elements in mind, I think it's better to revert the fix and remove the seed as this will probably cause more trouble than fixing things...

Unfortunately there is no way to use `SAVE_CONFIG` in Klippain until it's fixed on Klipper side...

Frix-x, 2023-04-15:
I'll also add a FAQ to explain this kind of things in the meantime

Surion79, 2023-04-25:
> Hey, hello!
> 
> So here is a post to sum up with some testings on my side:
> 
> 1. As of now, `SAVE_CONFIG` doesn't work natively with Klippain. It's a known thing of Klipper when the section to be saved is in a file included by `[include xxx.cfg]`
> 2. Putting the element you want to save in the `printer.cfg` like PIDs, make it work for this element only. If PIDs are defined in one of the included file, then we fall in case 1 and it fail. But this is not something conceivable in Klippain as there is some configuration included or not depending on every user (mainly on the Z offset settings).
> 3. `SAVE_CONFIG` will work for elements that doesn't need to be set (like a mesh) as the saved values will not need to "override" something in one of the includes (ie. we do not set a mesh points/values in our config)
> 4. Adding the seed helps to SAVE things without a bug. But if for some reason you have a let's say the PIDs set in the overrides.cfg file (as I'm doing currently), you will have a problem as the saved values will not be used (this could be confusing).
> 5. Last point, when using a new parameter from another thing like your beacon calibrate, it's like on point 3 if the element doesn't have a proper initialisation of its config. Or fall in the point 4 and need the seed to work.
> 
> So, with these elements in mind, I think it's better to revert the fix and remove the seed as this will probably cause more trouble than fixing things... EDIT: reverted in [51cefe4](https://github.com/Frix-x/klippain/commit/51cefe4448ddc6ff4ebc0046e918a8073c02b4ee)
> 
> Unfortunately there is no way to use `SAVE_CONFIG` in Klippain until it's fixed on Klipper side...

Regarding 4: wouldn't the auto generated part with PID under the include overrides.cfg  in the printer.cfg not triumph over settings in overrides.cfg?

i haven't looked but i guess there is already an issue on klipper github

Cisien, 2023-04-25:
Personally, I've always viewed the printer.cfg has holding the same role as overrides.cfg does in todays setup whenever i have ventured into multi-file configs. Its a bit hard to change now, nor am I suggesting that its required, but if you ever do a redesign of this, that may be something to consider

Surion79, 2023-04-25:
I think the pid topic is easy to fix by moving it from overrides to printer.cfg.
I don't know (yet) if z_offset is universally required for every configuration. If not, then it can't be placed centrally as default.

smwoodward, 2023-12-15:
I have discussed it more here 
https://github.com/Frix-x/klippain/pull/396
But the only thing that would be needed to make save_config work for both pid's and z_offset is to move those over to the printer.cfg file.  

Where the pids are easy to do as they are ONLY in the overrides.cfg file, the z_offset is in the readonly config files and can't be deleted from those files.  

If the z_offset line was removed from all of the probe config files and 

[probe]
z_offset: xxx

was added to printer.cfg, this would fix the issue.