Frix-x/klippain #137: Error evaluating 'homing_override:gcode':

kurtblah, 2023-04-06:
I get this error msg in the console upon start-up:
Error evaluating 'homing_override:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'sensorless_current_factor'

This happened after moving to klippain from your earlier version. Not sure where to look, the klippy log looks normal to me.

My machine is stock 2.4, with TAP and Stealthburner. normal endstops installed.

Frix-x, 2023-04-06:
Hi,

Thanks for opening this issue!
Yes this is due to a new variable added in the last version: `sensorless_current_factor`

You need to add it in your variables.cfg file and everything will be alright (see the variables.cfg templates for the definition and default value)

kurtblah, 2023-04-06:
thanks, that solved it.

Frix-x, 2023-04-07:
> adding to that seems like the sensorless homing variables need to be floats. I use sensorless homing on my machine and had to change those to floats in the overrides

This is good to know! I'm not using sensorless on my side and I tested it as far as I could when writing support for it. I'll change that in the next release :)

Tom-T, 2023-04-09:
I sent a patch for this as I had the same issue. :) I just winged it as I don't really know the language that well.

s0ybean, 2023-04-09:
I'm still having the same issue even after copying the latest template.  Not sure what it means to "change to floats" in the overrides :)

Can we just keep it simple and use a defined current rather than computing a percentage?

closed 2023-04-16 by Frix-x