Frix-x/klippain #140: Error evaluating 'homing_override:gcode': TypeError: can't multiply sequence by non-int of type 'float'

s0ybean, 2023-04-09:
Hi, 

I'm setting up a fresh install and i ran into this issue when i try to home:  !! Error evaluating 'homing_override:gcode': TypeError: can't multiply sequence by non-int of type 'float'

attached is the relevant klipper log

`Homing X
Error evaluating 'homing_override:gcode': TypeError: can't multiply sequence by non-int of type 'float'
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 61, in render
    return str(self.template.render(context))
  File "/home/pi/klippy-env/lib/python2.7/site-packages/jinja2/environment.py", line 1090, in render
    self.environment.handle_exception()
  File "/home/pi/klippy-env/lib/python2.7/site-packages/jinja2/environment.py", line 832, in handle_exception
    reraise(*rewrite_traceback_stack(source=source))
  File "<template>", line 80, in top-level template code
TypeError: can't multiply sequence by non-int of type 'float'
Error evaluating 'homing_override:gcode': TypeError: can't multiply sequence by non-int of type 'float'`

Frix-x, 2023-04-10:
Hi, thanks for raisin the issue. This is something already known and corrected in the develop branch that will be merged very soon. In the meantime, you can have a look at this commit for a fix: https://github.com/Frix-x/klippain/commit/94be381dcdfe974aa26210efe4a7841fc8e1dc0a

Frix-x, 2023-04-10:
I'll close it as it's also a duplicate of #137

closed 2023-04-10 by Frix-x