Frix-x/klippain #141: ERCF Happy Hare support

AMFkuna, 2023-04-11:
Hi Frix-x
Are you planning to support ERCF Happy Hare please?
https://github.com/moggieuk/ERCF-Software-V3

Thanks

Frix-x, 2023-04-11:
Hello!
Yes I'm working with @Benoitone to switch to Happy hare as the default ERCF system in Klippain. In the meantime you can already use it like that:

 1. Use the Happy Hare installer from moggieuk (`install.sh -i`) and let it do its job
 2. In your `printer.cfg` let only
```
[include ercf_hardware.cfg]
```
but remove the following lines:
```
[include client_macros.cfg]
[include ercf_software.cfg]
[include ercf_parameters.cfg]
```
  3. In the `ercf_hardware.cfg` file add to the top:"
```
[gcode_macro _USER_VARIABLES]
variable_ercf_enabled: True
gcode:

[include ercf_software.cfg]
[include ercf_parameters.cfg]
```
  4. Then in your `overrides.cfg` file, add this specific macro needed for the START_PRINT:
```
# Specific macro to raise an error during the print_start and stop in case of ERCF error
[gcode_macro _DO_ERROR_ERCF_PAUSE]
gcode:
    {% if printer.ercf.is_paused|int == 1 %}
        {action_raise_error("ERROR ERCF !!! UNLOCK ERCF FIRST and restart start_print macro.")}
    {% else %}
        {% if verbose %}
            RESPOND MSG="ERCF not paused"
        {% endif %}
    {% endif %}
```
  5. Check in the `ercf_hardware.cfg` file the pins and change them according to your `mcu.cfg`

With this, everything should work correctly. Also, you can be "safe" as the files are set in the "Klippain way" and will not be deleted on update :)

AMFkuna, 2023-04-11:
Many thanks for your great work !!

Surion79, 2023-04-25:
There is an issue. with this description, the Happy Hare will not boot up and throw a warning. The `[save_variables]` section in the `variables.cfg` points to apparently original ERCF variables. THis is incompatible with Happy Hare as the option `ercf_calib_clog_length = 8.0` is not set. this missing line errors out booting ERCF.

Benoitone, 2023-04-25:
Yes, in ercf_software.cfg
The 2 first lines must be comment like:

#[save_variables]
#filename: /home/pi/printer_data/config/ercf_vars.cfg

Surion79, 2023-04-25:
> Ho Yes, in ercf_software.cfg The first 2 lines must be commented like:
> 
> **#[save_variables] #filename: /home/pi/printer_data/config/ercf_vars.cfg**
> 
> So ercf use the default klippain variable file to save is variable which is defined at the end of klippain variable.cfg file(because only one save_variable file must be present.)
> 
> I don't know if it's necessary to add: **ercf_calib_clog_length = 8.0** in klippain save_variables.cfg ???

yes, as i described, you need to add the clog length or Happy Hare WILL NOT boot/start.

Benoitone, 2023-04-25:
I just try to remove `ercf_calib_clog_length` from my `save_variables.cfg` file and when i make the ercf calibration (`ERCF_CALIBRATE_SINGLE TOOL=0` or `ERCF_CALIBRATE`) `the ercf_calib_clog_length` has been created...

Surion79, 2023-04-26:
[image]
This doesn't work with the macro when I have an error... Not sure what the issue is yet

Benoitone, 2023-04-27:
OK, i just test too and I think it's a change in ERCF Happy Hare....

Benoitone, 2023-04-27:
I think it can work... Haven't been able to test yet...

```
# Specific macro to raise an error during the print_start and stop in case of ERCF error
[gcode_macro _DO_ERROR_ERCF_PAUSE]
gcode:
    {% if printer.ercf.tool|int != INITIAL_TOOL or printer.ercf.filament != "Loaded" %}
        {action_raise_error("ERROR in loading filament from ERCF !!! Check and restart the print.")}
    {% else %}
        {% if verbose %}
            RESPOND MSG="Filament load OK"
        {% endif %}
    {% endif %}
```

Frix-x, 2023-06-11:
Closing as Klippain v4 is merged and released :)

closed 2023-06-11 by Frix-x