Frix-x/klippain #142: Recommended WLED implementation method?

s0ybean, 2023-04-11:
Hi,

Just wondering if you have any ideas how an end user could implement WLED? All of the lighting calls are baked into the non-modifiable macros and uses a specific format. That makes it a bit hard to use overrides (from what I understand) since we wouldn't really need to specify rgb values.

WLED allows us to use complex animations and offload the processing to a very inexpensive esp32. All of the setup is done on the esp32 gui.

To use it in klipper, we simply add WLED to moonraker via IP or serial, and call the specific preset that we want (which was created through the WLED interface on the esp32).

For example:

WLED_ON string=chamber preset=1

This would load any complex animations or effects we set up for preset 1 without any extra configuring on the klipper side. 

Here is the documentation:
https://moonraker.readthedocs.io/en/latest/configuration/#wled

Frix-x, 2023-04-11:
There is currently no management of WLED in Klippain. But as everything is customization, my recommendation would be to override the `STATUS_LEDS` macro by copy pasting all its content in the overrides.cfg file and adding in it some logic to also callo `WLED_ON string=xxxx preset=yyy` as you want it.

s0ybean, 2023-04-11:
thank you for the reply! :)

So, a quick way to do this would be to add this to add this in override:

```

[gcode_macro WLED_ON]
description: Turn WLED strip on using optional preset and resets led colors
gcode:
  {% set strip = params.STRIP|default("chamber")|string %}
  {% set preset = params.PRESET|default(4)|int %}

  {action_call_remote_method("set_wled_state",
                             strip=strip,
                             state=True,
                             preset=preset)}

[gcode_macro STATUS_LEDS]
    {% if printer["gcode_macro _USER_VARIABLES"].status_leds_caselight_enabled %}
        WLED_ON STRIP="chamber" PRESET={status_color[color].caselight} TRANSMIT=1
    {% endif %}

```

and change this in variables:

```
'caselight': {
            'busy': {1},
            'cleaning': {2},
            'calibrating_z': {'3},
            'heating': {4},
            'homing': {5},
            'leveling': {6},
            'meshing': {7},
            'on': {8},
            'off': {9},
            'printing': {10},
            'standby': {11},
            'error': {12},
        },
```

am i on the right path?

s0ybean, 2023-04-13:
this is where I am so far.  I hope I am on the right track, but I am having a hard time figuring out how to get my WLED_ON macro to pull the number from my chamber variable and use it for preset.  I'm really bad with variables, especially since this original one is a bit tricky since it accomodates a lot of different setups

```
[gcode_macro _USER_VARIABLES]
variable_status_leds_chamber_enabled = True
variable_status_leds_chamber_led_name: "chamber"
gcode:
```

```
[gcode_macro WLED_ON]
description: Turn WLED strip on using optional preset and resets led colors
gcode:
  {% set leds_name = params.LEDS %}
  {% set color_name = params.COLOR %}
  {% set strip = params.STRIP|default("chamber")|string %}
  {% set preset = printer["gcode_macro _USER_VARIABLES"].status_leds_colors[leds_name][color_name] %}

  {action_call_remote_method("set_wled_state",
                             strip=strip,
                             state=True,
                             preset=preset)}
```

```
[gcode_macro STATUS_LEDS]
gcode:
    {% set color = params.COLOR|default('off')|lower %}
 
    {% set status_color = {
        'ready': {
            'logo': 'standby',
            'nozzle': 'standby',
            'caselight': 'standby',
            'chamber': 'standby',
            'minidisplay': 'on'
        },

*** i added "chamber" entry for every status color

    {% if printer["gcode_macro _USER_VARIABLES"].status_leds_chamber_enabled %}
        WLED_ON STRIP=chamber PRESET={status_color[color].chamber}
    {% endif %}
```

in variables I did this:

```
        },
        'chamber': {
            'busy': 1,
            'cleaning': 1,
            'calibrating_z': 1,
            'heating': 1,
            'homing': 1,
            'leveling': 1,
            'meshing': 1,
            'on': 1,
            'off': 1,
            'printing': 1,
            'standby': 1,
            'error': 1,
        },
```

Frix-x, 2023-04-15:
Hey how is it going? Did you managed to make it work?

I just had a look at your macros and it looks ok to me. At least it's how I would do it myself :)

s0ybean, 2023-04-15:
Hi! 

Unfortunately, it is not reading the variable I am using for preset.  Not sure if I need " or ' or {} around the single number?

s0ybean, 2023-04-16:
UndefinedError: dict object has no element Undefined

Surion79, 2023-05-03:
have you completed it or do you have still issues?

s0ybean, 2023-05-03:
i'm stuck.  was never able to figure out how to feed the variable properly

Surion79, 2023-05-03:
can you put here the whole picture? all gcode and config you use for the implementation? not only snippets

s0ybean, 2023-05-03:
to summarize -- the goal is to, for example, have     **STATUS_LEDS COLOR="PRINTING"** send the command **WLED_ON preset=1**,  **STATUS_LEDS COLOR="BUSY"** sends command **WLED_ON preset=2**, etc.

It should pick up the preset number from variables.cfg since i set a variable_status_leds_colors for WLED ('chamber'). For testing, all of my led colors are set to 1.

the documentation for using wled is located here https://moonraker.readthedocs.io/en/latest/configuration/#wled

If i test wled by sending manual commands or setting a startup preset, it works properly, so i definitely have a connection to it.

thank you!

attached are my variables.cfg and overrides.cfg files.
[overrides.txt](https://github.com/Frix-x/klippain/files/11391645/overrides.txt)
[variables.txt](https://github.com/Frix-x/klippain/files/11391646/variables.txt)

quoted are my relevant variables.cfg and overrides.cfg changes (excluding motor overrides) for quick reference:
```
################################
# WLED			               #
################################
[gcode_macro _USER_VARIABLES]
variable_status_leds_chamber_enabled = True
variable_status_leds_chamber_led_name: "chamber"
gcode:

[gcode_macro WLED_ON]
description: Turn WLED strip on using optional preset and resets led colors
gcode:
#  {% set leds_name 
    {% set leds_name = params.LEDS %}
    {% set color_name = params.COLOR %}
    {% set strip = params.STRIP|default('chamber')|lower %}
    {% set preset = printer["gcode_macro _USER_VARIABLES"].status_leds_colors[leds_name][color_name] %}
    

    {action_call_remote_method("set_wled_state",
                             strip=strip,
                             state=True,
                             preset=preset)}

[gcode_macro _SET_ALLLEDS_BY_NAME]
# set same color for all leds
gcode:
    {% set leds_name = params.LEDS %}
    {% set color_name = params.COLOR %}
    {% set led = printer["gcode_macro _USER_VARIABLES"]["status_leds_" + leds_name + "_led_name"] %}
    {% set color = printer["gcode_macro _USER_VARIABLES"].status_leds_colors[leds_name][color_name] %}
    {% set transmit = params.TRANSMIT|default(1) %}

    SET_LED LED={led} RED={color.r} GREEN={color.g} BLUE={color.b} WHITE={color.w}  TRANSMIT={transmit}
	
[gcode_macro STATUS_LEDS]
gcode:
    {% set color = params.COLOR|default('off')|lower %}
 
    {% set status_color = {
        'ready': {
            'logo': 'standby',
            'nozzle': 'standby',
            'caselight': 'standby',
            'chamber': 'standby',
            'minidisplay': 'on'
        },
        'busy': {
            'logo': 'busy',
            'nozzle': 'on',
            'caselight': 'busy',
            'chamber': 'busy',
            'minidisplay': 'on'
        },
        'heating': {
[... 166 lines cut ...]
            'leveling': 1,
            'meshing': 1,
            'on': 1,
            'off': 1,
            'printing': 1,
            'standby': 1,
            'error': 1,
        },
        'minidisplay': {
            'on': {'r': 0.0, 'g': 0.2, 'b': 0.4, 'w':1.0},
            'off': {'r': 0.0, 'g': 0.0, 'b': 0.0, 'w': 0.0},
            'error': {'r': 0.4, 'g': 0.0, 'b': 0.0, 'w':0.0},
        },         
        'thermal': {
            'hot': {'r': 1.0, 'g': 0.0, 'b': 0.0, 'w': 0.0},
            'cold': {'r': 0.3, 'g': 0.0, 'b': 0.3, 'w': 0.0}
        }
    }

## Do not remove the next lines
gcode:

[save_variables]
filename: ~/printer_data/config/save_variables.cfg
```

Frix-x, 2023-11-14:
I changed the title of this issue to track WLED but also Led_effect integration since it's highly linked and there is still currently no good implementation of this into Klippain...
This issue can also track rainbow barf integration as well

smwoodward, 2023-12-15:
I don't have anything for WLED integration, but I am working on BARF and I think I have it configured for the most part.  Would you want to create another branch for me to send the PR to to help look over it?

smwoodward, 2023-12-15:
https://github.com/Frix-x/klippain/pull/400

Surion79, 2023-12-16:
imho the pr needs major work done

Frix-x, 2024-02-26:
This is now merged on develop to be released soon :)

closed 2024-02-26 by Frix-x