Frix-x/klippain #144: Voron V0 implementation

fsaad1984, 2023-04-12:
It would be nice to have the Voron V0 implemented / supported. Many of us have one next to a V2.4 or Trident and it would be nice to have a similar config set on all printers.

Frix-x, 2023-04-15:
Hey, thanks for raising an issue.
I plan to add a real support for V0 in a future release. However in the meantime, you can already use Klippain (as I do on mine) by just setting the X,, Y and Z axis directly in your overrides. There is not su much more to do to make it work :)

Frix-x, 2023-04-15:
Ok, so I just added proper support for V0 printers in this commit: https://github.com/Frix-x/klippain/commit/f4359bc4a046b03357a28cc4709ab4519cd510aa
As I'm not running one currently, there could be some bugs. So please try it and let me know !

closed 2023-04-16 by Frix-x