Frix-x/klippain #146: caselights, with no color?

kurtblah, 2023-04-12:
My plain caselights do not turn on a print start. The LIGHT_OFF and LIGHT_ON macros work fine and are dim-able.

Where is the "is_light_enabled" var used in the PRINT_START macro set?

  {% if light_enabled %}
        LIGHT_ON S={light_intensity_start_print}
    {% endif %}

Frix-x, 2023-04-15:
Hey thanks for raising an issue.
Can you tell me more about how you configured your caselights? Because there is currently two ways:
  - Using FCOB white bars: then it's using the LIGHT_ON and LIGHT_OFF macros and the variable `light_enabled` is activated automatically
  - Using Neopixels leds bars: then it's using the STATUS_LEDS macros and the variable `status_leds_enabled` is activated automatically

I think you are running a hybrid way but I would like to understand how you've done it :)

kurtblah, 2023-04-15:
I have my caselights connected to a heater pin. (they're not neopixels).  That's it really, nothing fancy.

Frix-x, 2023-04-15:
So, just use the FCOB configuration. They are working exactly like that :)

kurtblah, 2023-04-15:
That's what I am using, but the caselights never went on during a print.  I needed to use the LIGHT_ON macro to turn them on after a print started.  Not really a big deal, but Once I added the line to the variables.cfg it seemed to work correctly.

I will re-trace my steps to make sure this is how it is.

kurtblah, 2023-04-15:
I think my issue was I was using floats (0 - 1.0) instead of 0 -100 for the light_intensity vars. My mistake. The added line is not needed.

Frix-x, 2023-04-15:
> I think my issue was I was using floats (0 - 1.0) instead of 0 -100 for the light_intensity vars. My mistake. The added line is not needed.

Ohhh ok, this explain it and is definitely a bug. I'll correct that :)

closed 2023-04-16 by Frix-x