Frix-x/klippain #147: Can not save any changes to files 'installed' by your Klippain system.

zeplintwo, 2023-04-13:
Any changes made to the core files that are uploaded by github are being blocked from the 'update_manager Klippain' 

below is the log file from moonraker showing the forbiden access to your files. this is not a good thing.

`2023-04-13 06:18:22,577 [app.py:log_request()] - 201 POST /server/files/upload (192.168.1.144) [_TRUSTED_USER_] 12.41ms
2023-04-13 06:18:23,676 [klippy_connection.py:_on_connection_closed()] - Klippy Connection Removed
2023-04-13 06:18:23,930 [klippy_connection.py:_do_connect()] - Klippy Connection Established
2023-04-13 06:18:29,646 [klippy_connection.py:_request_initial_subscriptions()] - Webhooks Subscribed
2023-04-13 06:18:29,648 [klippy_connection.py:_request_initial_subscriptions()] - GCode Output Subscribed
2023-04-13 06:18:29,651 [app.py:register_remote_handler()] - Registering HTTP endpoint: (GET POST) /printer/motion_report/dump_trapq
2023-04-13 06:18:29,652 [websockets.py:register_api_handler()] - Registering Websocket JSON-RPC methods: printer.motion_report.dump_trapq
2023-04-13 06:18:29,910 [job_state.py:_handle_started()] - Job state initialized: standby
2023-04-13 06:18:30,171 [klippy_connection.py:_check_ready()] - Klippy ready
2023-04-13 06:18:30,669 [data_store.py:_init_sensors()] - Configuring available sensors: ['heater_bed', 'temperature_sensor RaspberryPi', 'temperature_sensor mcu', 'temperature_sensor toolhead_mcu', 'extruder']
2023-04-13 06:18:51,385 [web.py:log_exception()] - 403 POST /server/files/upload (192.168.1.144): Access to file controller_fan.cfg forbidden by reserved path 'update_manager Klippain'
2023-04-13 06:18:51,427 [app.py:log_request()] - 403 POST /server/files/upload (192.168.1.144) [_TRUSTED_USER_] 55.31ms
2023-04-13 06:18:55,420 [web.py:log_exception()] - 403 POST /server/files/upload (192.168.1.144): Access to file controller_fan.cfg forbidden by reserved path 'update_manager Klippain'
2023-04-13 06:18:55,427 [app.py:log_request()] - 403 POST /server/files/upload (192.168.1.144) [_TRUSTED_USER_] 19.48ms
2023-04-13 06:19:05,247 [app.py:log_request()] - 201 POST /server/files/upload (192.168.1.144) [_TRUSTED_USER_] 25.72ms
2023-04-13 06:19:17,941 [web.py:log_exception()] - 403 POST /server/files/upload (192.168.1.144): Access to file mainsail.conf forbidden by reserved path 'update_manager Klippain'
2023-04-13 06:19:17,948 [app.py:log_request()] - 403 POST /server/files/upload (192.168.1.144) [_TRUSTED_USER_] 19.52ms
2023-04-13 06:19:26,740 [web.py:log_exception()] - 403 POST /server/files/upload (192.168.1.144): Access to file mainsail.conf forbidden by reserved path 'update_manager Klippain'
2023-04-13 06:19:26,747 [app.py:log_request()] - 403 POST /server/files/upload (192.168.1.144) [_TRUSTED_USER_] 21.08ms
2023-04-13 06:21:12,136 [web.py:log_exception()] - 403 POST /server/files/upload (192.168.1.144): Access to file mainsail.conf forbidden by reserved path 'update_manager Klippain'
2023-04-13 06:21:12,144 [app.py:log_request()] - 403 POST /server/files/upload (192.168.1.144) [_TRUSTED_USER_] 21.16ms
2023-04-13 06:21:16,616 [web.py:log_exception()] - 403 POST /server/files/upload (192.168.1.144): Access to file mainsail.conf forbidden by reserved path 'update_manager Klippain'
2023-04-13 06:21:16,623 [app.py:log_request()] - 403 POST /server/files/upload (192.168.1.144) [_TRUSTED_USER_] 21.18ms`

Benoitone, 2023-04-13:
hello, you must use overrides to make modifications or add your own functions
https://github.com/Frix-x/klippain/blob/main/docs/overrides.md

Frix-x, 2023-04-15:
Hey,

As Benoitone have already mentionned, for the update manager to work, you should not modify any files from my repo: it need to be kept read-only. This is not really a limitation of my repository but how the update manager system is working.

So that's why there is the "override" mechanism in Klippain. If you need to change something: do not modify it directly, but add an override to "replace" it in your own user `overrides.cfg` file :)