Frix-x/klippain #150: modular start_print issue

jeronbernal, 2023-04-16:
getting a error when trying to add the primeline to my start print via variable_startprint_actions: "primeline"
	
	i get a error of Error evaluating 'gcode_macro START_PRINT:gcode': gcode.CommandError: Unknown module called in START_PRINT! Please verify your startprint_actions variable override!
[image]
[image]

jeronbernal, 2023-04-16:
tried to use 3.1.0 and put it in the override.cfg but get this issue 

[image]
[image]

Benoitone, 2023-04-16:
Maybe you can't use only primeline ?

Frix-x, 2023-04-16:
Yes Benoitone is right. This variable is expected to be a list of action and in Jinja it need at least two elements.

@Benoitone, I'm already discussing with him in the Klippain channel of the Voron Discord. So I'll close this issue as the solution is found :)

closed 2023-04-16 by Frix-x