Frix-x/klippain #154: MCU_UARTS not defined on BTT_SKR_E3mini_v2.cfg

AMFkuna, 2023-04-24:
Hi 
I can implement Klippain to my V0 printer, but mcu uarts not defined in BTT_SKR_E3mini_v2.cfg

I manualy update my mcu.cfg with correct pins.

Frix-x, 2023-04-25:
The thing is that there is no way to "remove" something already declared. So if I declare by default the UART in the V0 axis cfg, then all the user of other board will not be able to use them as they will not be able to remove it.

However the opposite is really easy: it's done using the overrides.cfg file. So, in your case you need to:
  1. enable the v0 files from printer.cfg
  2. add the uart definition to your overrides that way:
```
[tmc2209 stepper_x]
uart_pin: MCU_TMCUART
tx_pin: MCU_TMCTX
uart_address: 0

[tmc2209 stepper_y]
uart_pin: MCU_TMCUART
tx_pin: MCU_TMCTX
uart_address: 2

[tmc2209 stepper_z]
uart_pin: MCU_TMCUART
tx_pin: MCU_TMCTX
uart_address: 1
```

AMFkuna, 2023-04-25:
Yes I did that, but in mcu_defaults there must not be defined uarts X_TMCUART=MCU_XM_UART for BTT_SKR_E3mini
[image]

Surion79, 2023-04-30:
yeah, I have a local branch for the SKR Pico which has the same type of configuration. that does not work (anymore?)

Frix-x, 2023-06-11:
Closing as Klippain v4 is merged and released :)

closed 2023-06-11 by Frix-x