Frix-x/klippain #159: Adding different TMCs in Klippain

Surion79, 2023-04-25:
I open this issue for discussing, how this could be implemented. ( If this comes to an agreement, I offer to create a PR based on the conclusion of this issue)

Current situation: Due to the restrictions on Klipper, the tmc types must be defined directly.
Challenge: Apart the "default" 2209, BTT also added a TMC2240 for extruder in their new SB2240 toolhead pcb. The v24 uses TMC5160s. So, different TMCs coming around.

In general, the configuration betweet stepper and tmc must be separated to be able to be included individually.

Approach 1: (generic, all tmc sections)
For each type in X, Y, Z (Z as group, not individual Zs) and E create a dedicated tmc file, which can be included in the printer.cfg individually.
Default could be added for X, Y, Z, E x 2209, 2240, 5160 (more can be added easily via a new file)

Approach 2: (only E tmc sections)
Instead all types, only E is set up for different TMC types. If a toolhead PCB is used, it would mean that configs e.g. between SB2209 and SB2240 would differ mainly in the part of the tmc config (UART vs SPI). (And these refer to the e_tmc2209.cfg, so it can be included in the board only sections, if no toolhead is used.)

This would make the setup easier, since you select the appropriated toolhead pcb, including the version and don't care about the background TMC. e.g. BTT_EBB_SB2240_v1.0.cfg or BTT_EBB_SB2209_v1.0.cfg

@Frix-x which is your prefer approach? Do you have another option?

Surion79, 2023-04-25:
Yeah, i saw the SB2209/SB2240 PR, but i think the combination of SB2209 and SB2240 in that PR is somewhat complexing. 

would that mean that you have something like <printertype>_<degree>_<Type>_<axis>_<tmc>.cfg naming?
You multiply then other parameters with tmc types, which can become very large, since there is also a lot of duplication of config?
PrinterType * Degree * Type * Axis * Tmc for Z. XY is "shorter", but also includes "V0Specific"

I compared the Z configs and the difference is "only" in the stepper section, not in the tmc.
1.8d vs 0.9d makes sense to divide it, since the difference is in the same section.

Frix-x, 2023-04-25:
Yes you are right on this point. I need to think about it a little bit more as I don't have enough perspective to be able to make a decision right now.

If you want to work on that, feel free to issue a draft PR with your idea and let's iterate on it :)

Surion79, 2023-04-25:
PR is posted :)

M3HNGRY, 2023-04-26:
What about the TMC2226? It's suppose to be exact like the TMC 2209 but with sense_resistor: 0.150

Surion79, 2023-04-26:
The listed tmcs were examples...

Frix-x, 2023-05-04:
The base of this was added in https://github.com/Frix-x/klippain/commit/333d802f2fe6108a3f03b12d540f68c20bc66c74
Now we need to add support for other type of TMCs

closed 2023-06-11 by Frix-x