Frix-x/klippain #164: Rainbow BARF LED Toolhead Support for StealthBurner

M3HNGRY, 2023-04-26:
Would like to request support for this if possible. 

[neopixel sb_leds]
pin: PB0
# The pin connected to the neopixel. This parameter must be provided.
chain_count: 10
# 8 Rainbow barf + 2 Toolhead
# The number of Neopixel chips that are "daisy chained" to the
# provided pin. The default is 1 (which indicates only a single
# Neopixel is connected to the pin).
color_order: GRB,GRB,GRB,GRB,GRB,GRB,GRB,GRB,GRBW,GRBW
# Set the pixel order required by the LED hardware. Options are GRB,
# RGB, GRBW, or RGBW. The default is GRB.
initial_RED: 1.0
initial_GREEN: 0.45
initial_BLUE: 0.0
initial_WHITE: 0.0
# Sets the initial LED color of the Neopixel. Each value should be
# between 0.0 and 1.0. The WHITE option is only available on RGBW
# LEDs. The default for each color is 0.#

V/R
Anthony

Surion79, 2023-04-26:
this is based on a different repo for led_effect. it is more an integration request for led effect instead of just a barf led?

M3HNGRY, 2023-04-26:
Both it has the setting for the CENTER LOGO LED and coding for the effects. I was looking at the status_led and its similar.

Surion79, 2023-04-27:
it seems that it possible to place it in the overrides.cfg for the time being, until it is clear when and how it is integrated.

LongDono, 2023-08-31:
For anyone wondering, this is from https://github.com/VoronDesign/Voron-Stealthburner/blob/main/Firmware/stealthburner_leds.cfg

W141-ID, 2023-09-06:
No update here?

mrstone14, 2023-10-30:
@W141-ID  could you elaborate where you are placing that example code to apply the led_effect?

W141-ID, 2023-11-03:
> @W141-ID could you elaborate where you are placing that example code to apply the led_effect?

I have added my costum status_leds config  below. I uploaded it to the the config directory of my printes klippain folder (same folder as where the printer.cfg is stored) and included the file via the printer.cfg

If there is intrest behind this implementation I could work on a viable solution.

```
# This file is automatically included when status_leds hardware config file is included

# src : https://github.com/VoronDesign/Voron-Stealthburner/blob/main/Firmware/stealthburner_leds.cfg
# Contributed by Voron discord users wile.e, Tetsunosuke, and etherwalker
# Heavily modified by Frix-x to be more adapted for the generic config goal

# The following status macro is available:
#    STATUS_LEDS COLOR="xxxxx"

# With additional macros for basic control:
#    SET_NOZZLE_LEDS_ON
#    SET_LOGO_LEDS_OFF
#    SET_NOZZLE_LEDS_OFF

#-------------------------#
#    Rainbow neopixel     #
#-------------------------#

[gcode_macro _USER_VARIABLES]
variable_status_leds_logo_idx: '1,2,3,4,5,6,7,8'
variable_status_leds_nozzle_idx: '9,10'

[neopixel status_leds]
pin: toolhead:STATUS_NEOPIXEL
#   The pin connected to the neopixel. This parameter must be provided.
chain_count: 10
#   The number of Neopixel chips that are "daisy chained" to the
#   provided pin. The default is 1 (which indicates only a single
#   Neopixel is connected to the pin).
color_order: GRB, GRB, GRB, GRB, GRB, GRB, GRB, GRB, GRBW,  GRBW
#   Set the pixel order required by the LED hardware. Options are GRB,
#   RGB, GRBW, or RGBW. The default is GRB.
initial_RED: 1.0
initial_GREEN: 0.0
initial_BLUE: 1.0
initial_WHITE: 0.0
#   Sets the initial LED color of the Neopixel. Each value should be
#   between 0.0 and 1.0. The WHITE option is only available on RGBW
#   LEDs. The default for each color is 0.#

##########################
# LED Effects Animations #
##########################

##################
## logo effects ##
##################

[led_effect sb_logo_busy]
autostart:              false
frame_rate:             24
leds:
    neopixel:status_leds (1-8)
layers:
    breathing  3 1 top (1,0,0)

[led_effect sb_logo_cleaning]
autostart:              false
frame_rate:             24
leds:
    neopixel:status_leds (1-8)
layers:
    breathing  3 1 top (0.0, 0.02, 0.5)
[... 375 lines cut ...]
    {% endif %}

    {% if color == 'error' %}
        STOP_LED_EFFECTS
        SET_LED_EFFECT EFFECT=sb_critical_error
        set_nozzle_leds_on
    {% endif %}

    {% if not (color in status_color) %}
        { action_raise_error("COLOR is not valid!") }
    {% endif %}

    {% if printer["gcode_macro _USER_VARIABLES"].status_leds_enabled %}
        _SET_LEDS_BY_NAME LEDS="logo" COLOR={status_color[color].logo} TRANSMIT=0
        _SET_LEDS_BY_NAME LEDS="nozzle" COLOR={status_color[color].nozzle} TRANSMIT=1
    {% endif %}

    {% if printer["gcode_macro _USER_VARIABLES"].status_leds_caselight_enabled %}
        _SET_ALLLEDS_BY_NAME LEDS="caselight" COLOR={status_color[color].caselight} TRANSMIT=1
    {% endif %}

    {% if printer["gcode_macro _USER_VARIABLES"].status_leds_minidisplay_enabled %}
        _SET_ALLLEDS_BY_NAME LEDS="minidisplay" COLOR={status_color[color].minidisplay} TRANSMIT=1
    {% endif %}
```

Frix-x, 2023-11-14:
This issue is highly linked to #142 (just for reference and to track it for a next version)

pireybr, 2023-12-23:
> # This file is automatically included when status_leds hardware config file is included
>
> # src : https://github.com/VoronDesign/Voron-Stealthburner/blob/main/Firmware/stealthburner_leds.cfg
> # [...]

I'm getting this error, can you help me?

Section 'led_effect sb_logo_busy' is not a valid config section

Frix-x, 2024-02-22:
Please check out the development branch, a lot of work was done on the LED system a few weeks ago and led_effect support is already there as well as barf support, etc. The release is hopefully planned for tomorrow :)

closed 2024-02-26 by Frix-x