Frix-x/klippain #168: _MODULE_PRIMELINE not working with small bed sizes

Surion79, 2023-04-28:
Due to the Prime Line Length, the prime line would always overextend on a v0 since it has Y+80 and Y+130 commands, resulting in an out of bounce error

Surion79, 2023-04-29:
idea: if the original primeline length exceeds bed size, the primeline and its extrusion is divided so it can fit on the print bed and not run out

Frix-x, 2023-04-29:
In fact it doesn't need to be that long. I'll see if I can just change it to something more overextruded over 3-4cm only.

Surion79, 2023-04-29:
that works also. could also make it customizable with a variable "primelinelength", if some users want to extrude more

Frix-x, 2023-04-30:
I've implemented something a little bit better and more generic as it's now using the max_extrude_cross_section to compute its speed in order to avoid an error if the user set a flow rate very high.
Flow rate is configurable as well. I still need to do a littlbe bit more testing but it should work in most cases.

See commit https://github.com/Frix-x/klippain/commit/ed3e7d4a84924c5b7106492602e780f645e70bfe

yanfoo, 2023-04-30:
I have overridden many of Pain's gcode including the _MODULE_PRIMELINE. I use the adaptive Voron Purge in its place, it can be found at KAMP here https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging

I found it is easier to remove because it is thicker and uses very little space on the bed.

Surion79, 2023-04-30:
Meh, part of KAMP uses object exclusion which is afaik not compatible with ercf.

Frix-x, 2023-04-30:
Yes, my new prime line system is very similar to what you can find in KAMP : much more thick and smaller to be able to put it in all bed size. Also there is now a parameter for flowrate and one for length. However it's static and not an "adaptive" purge as I think there is no point of doing this...

[image]

Also on an other topic, I agree with Surion79. Using exclude object tags is not really good as less precise than the SIZE parameter and it does not contain the purge block that is used in case of using MMU systems such as the ERCF. So that's why I keep my own adaptive bed mesh for now.

yanfoo, 2023-05-03:
Very nice, I will try it.
Continue ton bon travail!

closed 2023-06-11 by Frix-x