Frix-x/klippain #174: Steper Z poition endstop is not valid

22MarioZ, 2023-05-02:
Hello

I'm using klippain for Trident configuration and I'm geting Klipper error for the following configuration: 
Option 'position_endstop' is not valid in section 'stepper_z'. 

Looks that issue is for the files: [Trident_1.8d_TR8x4_z.cfg](https://github.com/Frix-x/klippain/blob/main/config/hardware/axis/Z/Trident_1.8d_TR8x4_z.cfg) and [Trident_1.8d_TR8x8_z.cfg](https://github.com/Frix-x/klippain/blob/main/config/hardware/axis/Z/Trident_1.8d_TR8x8_z.cfg).

However if I'm using the same file on the root of the config folder (with necessary changes to include right files) I don't get any error.

Frix-x, 2023-05-02:
Oh indeed, I think you are using TAP ? If this is the case, it's definitely a bug that should be solved in the next release :)

Frix-x, 2023-05-02:
Fixed in https://github.com/Frix-x/klippain/commit/670551ac077bf20addffb831938a8d12a7e7e1de

closed 2023-06-11 by Frix-x

22MarioZ, 2023-06-21:
I need to reopen this ticket as the bug was re-introduced with 4.0.1. Can you fix it soon?

22MarioZ, 2023-06-21:
Ignore my previous comment...