Frix-x/klippain #175: Moonraker can't connect to Klipper! Check klippy_uds_address.

22MarioZ, 2023-05-02:
Hello,

I get following error when making configuration using klippain:

Moonraker can't connect to Klipper!

Please check if the Klipper service is running and klippy_uds_address is correctly configured in the moonraker.conf.

If I add new section to moonraker.conf:

```
[server]
klippy_uds_address: /tmp/klippy_uds
```

works without error. The configuration is from default Mainsail moonraker config.

Frix-x, 2023-05-02:
Hello, I think you are using the old Moonraker config paths (from before november 2022). This is not compatible with Klippain.

Just use KIAUH to uninstall Klipper and Moonraker, then use it again to reinstall Klipper and Moonraker. This way you will use the last directory structure and be future proof (as well as compatible with Klippain).
I think I should add a note about that in the documentation :)

22MarioZ, 2023-05-02:
Thank you. After upgrading following your instructions and reinstalling klippain start to work without error.

Frix-x, 2023-05-03:
Super ! Then problem solved and I'll close this issue :)

closed 2023-05-03 by Frix-x