Frix-x/klippain #197: Input Shaper workflow for seperate Axes

thomasfjen, 2023-05-10:
Hey would it be possbile to have the workflow seperated for the different Axes for bed slingers.
I am using a Voron Switchwire conversion and i am only using one ADXL which i place at the toolhead and bed separately.

Would be nice if the process / macro of belt tension and input shaping could be started for only one axis.

Frix-x, 2023-05-10:
Hello, yes this is perfectly doable. I'll see if I can work on something after the V4 is released in the following days :)

Fragmon, 2023-06-08:
you can add my edit:

[gcode_macro AXES_SHAPER_CALIBRATION]
description: Run standard input shaper test for all axes
gcode:
    {% set verbose = params.VERBOSE|default(true) %}
    {% set axes = params.AXES|default("x,y,xy")|string|lower %}
    {% set max_x = printer.toolhead.axis_maximum.x|float %}
    {% set max_y = printer.toolhead.axis_maximum.y|float %}

    {% if axes=="xy" %}
        TEST_RESONANCES AXIS=X
        M400
        TEST_RESONANCES AXIS=Y
        M400   
        {% if verbose %}
        RESPOND MSG="Graphs generation... Please wait a minute or two and look in the configured folder."
        {% endif %}
        RUN_SHELL_COMMAND CMD=plot_graph PARAMS=SHAPER
    
    {% elif axes=="y" %}
        TEST_RESONANCES AXIS=Y
        M400     
        {% if verbose %}
        RESPOND MSG="Graphs generation... Please wait a minute or two and look in the configured folder."
        {% endif %}
        RUN_SHELL_COMMAND CMD=plot_graph PARAMS=SHAPER

    {% elif axes=="x" %}
        TEST_RESONANCES AXIS=X
        M400     
        {% if verbose %}
        RESPOND MSG="Graphs generation... Please wait a minute or two and look in the configured folder."
        {% endif %}
        RUN_SHELL_COMMAND CMD=plot_graph PARAMS=SHAPER

    {% else %}
        {action_respond_info("Please choose the axis: x, y or xy")}
    {% endif %}

Frix-x, 2023-10-08:
This is now done alongside the new input shaper workflow and scripts that will be available in the following days (next Klippain release) :)

thomasfjen, 2023-10-08:
This is great! Thank you very much!

closed 2023-10-08 by thomasfjen