Frix-x/klippain #198: Rework of the ERCF integration

Frix-x, 2023-05-10:
Hi @Benoitone, @elpopo-eng and @Surion79 :)

As there was some work done by each of you in Klippain regarding the ERCF, I'm opening this issue to synchronise things a little bit.
Now that Ette ERCF system is not evolving more and that moggieuk took the lead on the future of ERCF with Ette approval, I would like to fully rework the ERCF Klippain's integration to make it more aligned with the current state of Klippain and be more future proof :)

1. First thing to do is to remove all the current ERCF macros and integrations to avoid getting duplicate and complicated mechanisms to handle both at the same time.
1. Then as a first step of developpement, I eard that HH have an install script and that everything should "almost just work correctly". So we could add a dummy config file included in the printer.cfg for ERCF. This file would set the internal Klippain "ercf_enabled" variable. Then in the startup check macro, we could check for "if there is ercf_enabled, then check if there is also the HH printer.ercf.enabled variable set". If it's not the case, then error out to stop Klipper start and warn the user that to use an ERCF he should install HH by himslef.
1. Then we should add all the things needed to make it work better in Klippain's macros as the START_PRINT things, etc...

Surion79, 2023-05-10:
@Frix-x do you want to create a branch for us regarding ercf integration? So we can PR to it independently to hypercare of v4

Benoitone, 2023-05-10:
> @Frix-x do you want to create a branch for us regarding ercf integration? So we can PR to it independently to hypercare of v4

maybe it's a good idea...

Benoitone, 2023-05-10:
> What kind of ercf are you using? I am using unmodded ercf with mellow fly can board and HH 1.2.4 with HH Klipperscreen. Buffer solution was ercp but that is not reliable for me. Looking currently at pika and the quiver

i actualy use unmodded v1.1 ercf too (except the encoder... i use the Binky Encoder) with Tircow easy board ,HH and HH Klipperscreen too. I use ERCP buffer because i have them from my first v1.0 ERCF...

Frix-x, 2023-05-10:
Ok, I'll create this branch based on develop.
I think the work is not that hard and could be done in a couple of days. So I'll wait for it to be merged in v4 before the release. In the meantime, I'll work myself on the Swithcwire and multikinematic PR :)

Frix-x, 2023-05-10:
Feel free to PR on https://github.com/Frix-x/klippain/tree/ercf-rework 
I'll see if I can give you some contributer right to let you push directly to the branch without using your fork : it will be more efficient :)

Surion79, 2023-05-10:
i commented all ercf macro calls with ### since most of them are kinda required in the end and we can verify type and placement of them more easily if we have a "visual" anchor. I reworked the ercf.cfg already and removed the ercf macro file. 
My currently idea is to use the output files of HappyHare like installed via install.sh in root folder for now. i do hope that it is possible to place such "huge" mod like ERCF in a subfolder in the config folder of klipper. but this would only work if the install script of HH is modified. Another topic for later :D just that you understand how i approach this

Benoitone, 2023-05-10:
The include you put in ercf.cfg will not work because of the symlink 🥲 you must add them in a file in printer_data like printer.cfg that will unfortunately not be updated automatically.

Surion79, 2023-05-10:
damn, thats sad! :D

Surion79, 2023-05-10:
and if we symlink them in klippain to the ercf folder? would that work?

Surion79, 2023-05-10:
in the branch i commented everything of ercf out and "reworked" it. stuff i don't understand, i kept commented out, so this should be discussed.

Surion79, 2023-05-10:
so the commits can be discussed individually

Surion79, 2023-05-10:
apart of the symlink issue we should discuss possible workarounds, the rest should be sufficient to run ercf HH software IMO. (but i guess the devil is in the detail :) )

Frix-x, 2023-06-11:
Closing as Klippain v4 is merged and released :)

closed 2023-06-11 by Frix-x