Frix-x/klippain #201: Adding bed fan functionality

Surion79, 2023-05-11:
Adding bed fan functionality for chamber soaking with automated speeding similar provided by Ellis.
Just as reminder.

Frix-x, 2023-05-13:
On one of my machine, I'm using beds_fan and run them by just selecting the nevermore config file. What functionnality do you miss?

Surion79, 2023-05-13:
the scripts provided by ellis do temperature based fan control. it is completely independant from the start_filter / stop_filter macros for the nevermore. Also i'd like to keep it generic. not everone has nevermore. Some have 2,3 or 4 bed fans and no nevermore, i have a nevermore v5 and want to add 1 fan on each side. So nevermore and 2 fans.
Currently this is just a design for 1 fan control, meaning, you need to Y slice the cables. otherwise, it would be a bit more complicated :) (I guess we would need to have a dedicated config for each number of bedfans if controlled individually)

Surion79, 2023-05-13:
the fan control works only good, if a chamber thermistor is used. for chamber soak. so my question was, if this would work with your chamber soak

W141-ID, 2023-05-14:
I was about to open op the same issue.

Today I installed two bed fans additional to nevermore. Had to tinker the filter macro a little bit but works fine.

Would be awesome to have a user variable asking for bed fans.

Also in the heat_soak macro the fans stop as soon as the required chamber temperature has been achieved. I noticed that as soon as this happens the chamber temperature drops a couple of degrees. It would be more effective to have it run all the time to have higher and consistent chamber temperatures.

Benoitone, 2023-05-14:
> Also in the heat_soak macro the fans stop as soon as the required chamber temperature has been achieved. I noticed that as soon as this happens the chamber temperature drops a couple of degrees. It would be more effective to have it run all the time to have higher and consistent chamber temperatures.
> 
> That's another reason to have a separate bed fans macro similar to Ellis just more generic.

I don't have a bed fan but I also noted that nevermore stopped at the end of _MODULE_HEATSOAK_CHAMBER and I didn't really see the reason, especially now with the new material management? In my case, I removed the STOP_FILTER from the HEATSOAK_CHAMBER module. I had thought of making an issue out at the time to ask the question to @Frix-x but I forgot. This is an opportunity to do so.

Surion79, 2023-05-14:
> I don't have a bed fan but I also noted that nevermore stopped at the end of _MODULE_HEATSOAK_CHAMBER and I didn't really see the reason, especially now with the new material management? In my case, [...]

well, heatsoak chamber-Filterspeed=1, startprint-filterspeed = materialbased speed
So it is ok imho to stop filter after the module heatsoak chamber since it is a module.

W141-ID, 2023-05-14:
> well, heatsoak chamber-Filterspeed=1, startprint-filterspeed = materialbased speed
> 
> So it is ok imho to stop filter after the module heatsoak chamber since it is a module.

As that might be right it would have been better to let it run anyways as the chamber temperature drops as soon as the fans stop.

A solution would be instead of stopping, changing the speed to the material speed.

I also don't like that the fans only start when the bed has heated up it saves a lot of time if they start simultaneously.

Surion79, 2023-05-14:
could we please split this issue in two seperate ones? it is now being hijacked regarding the stop_filter issue. because it is turned off for 15(?) seconds

W141-ID, 2023-05-14:
Don't see why it needs to be split. It is regarding the bed fan functionality.

Splitting it would make any meaningful implementation and improvements unnecessary difficult.

Surion79, 2023-05-14:
> Don't see why it needs to be split. It is regarding the bed fan functionality.
> 
> Splitting it would make any meaningful implementation and improvements unnecessary difficult.

Klippain is a generic configuration for a wide set of printers. this includes printers with bedfans without filter and filter without bedfans and printers with neither or both.