Frix-x/klippain #202: Adding KNOMI screen compatibility

Surion79, 2023-05-11:
WIP ⚠️

Frix-x, 2023-06-11:
I'll close this PR as it look like there is nothing to be added to Klippain for now and it should just work.
There is still the macro addition problem, but it's a problem on the KNOMI side, not in Klippain.

We'll reopen this issue if it needs more work but I'm pretty sure it will be ok :)

closed 2023-06-11 by Frix-x

jace, 2024-08-06:
I just installed a Knomi 2 and it's stuck on the G28 macro override conflicting with `homing_override`. @Frix-x Knomi's requirements seem trivial enough to include support in Klippain. It just requires setting macro variables and doesn't look like it'll have any unwanted side effects even if always enabled.