Frix-x/klippain #207: Existing command 'M190' not found in gcode_macro rename

ZiiB, 2023-05-15:
Heatbed related maccros are not instanciated by Klipper if printer config does not mention a heatbed. 
Therefore getting `Existing command 'M190' not found in gcode_macro rename` on startup if `[include config/hardware/heated_bed.cfg]` is not uncommented in printer.cfg

Frix-x, 2023-05-15:
Thanks I'll correct that quickly :)

Surion79, 2023-05-15:
> Thanks I'll correct that quickly :)

something for the startup?

Frix-x, 2023-05-16:
No I think it will be better to include the heater control fix (https://github.com/Frix-x/klippain/blob/main/macros/helpers/heaters_ctrl.cfg) directly from the heater_bed.cfg and extruders cfg files. This way we will not have any problem like this one. Then, by adding some checks to add to the M190 and other heat macros to validate that the section is included before calling it every time, will allow Klippain to work without an heater_bed defined (as ZiiB tried to do).

Frix-x, 2023-05-22:
This is fixed in the last commit: https://github.com/Frix-x/klippain/pull/188/commits/f20f395cdc747a80735daf7d0bcc7049f1374518

Frix-x, 2023-06-11:
Closing as Klippain v4 is merged and released :)

closed 2023-06-11 by Frix-x