Frix-x/klippain #210: Chamber heatsoak showing "incorrect" OK even when temperature is not met

Surion79, 2023-05-19:
Just installed a chamber thermistor and it was strange that the temperature was showed as "OK" but it was not. There was no info like bed soak that the timer has passed. 
It should be a bit more precise about this.

Benoitone, 2023-05-19:
Maybe the value of variable_print_default_chamber_max_heating_time: ?

Surion79, 2023-05-22:
well, in my test it was only an OK, without reaching the time limit. there was no info about "maximum time elapsed" or something

Frix-x, 2023-05-23:
If OK is printed in the console, it means that the chamber temperature is already above the temperature setpoint (set from the START_PRINT `CHAMBER` parameter or defaulting to `variable_print_default_chamber_max_heating_time`).

Benoitone, 2023-05-24:
So maybe change the RESPOND MSG="Chamber temperature OK !"
To something like
RESPOND MSG="Chamber temperature OK or maxtime reached !" ?

Frix-x, 2023-06-11:
I think Benoitone is right. I just understand the problem. We'll add something like that to the next version :)

Surion79, 2023-06-11:
i would like to be more specific and not generic about what could have happened

closed 2023-08-18 by Frix-x