Frix-x/klippain #216: Add Beacon surface scanner support

maskim13, 2023-05-27:
https://docs.beacon3d.com/quickstart/

Frix-x, 2023-05-28:
I know that some Klippain users are already using it without any problem. They installed the plugin and added an override with the beacon definition and that's it :)

What do you miss ?

Surion79, 2023-05-28:
A little bit more text would be nice to understand what your challenge with this is. Nearly everything can be done via overrides.
There are tools, that Klippain integrates, but still require additional installations and activities like ERCF or Z Auto Calibrate.

maskim13, 2023-05-28:
I did have issues initially due to sensorless homing (modified for TMC5160s). 

I was planning to use auto Z calibration with a physical endstop (which was already working) for Z homing, and use beacon for QGL and bed mesh, but encountered errors complaining about different things; Virtual endstops not being compatible with something else that was enabled, safe Z home (recommended for Beacon) not working with homing override, and possibly some others issues which I've forgotten. 

Beacon is currently working with homing override, without safe Z home, but may need set_kinematic_position used to prevent some complications.

With Beacon working, the physical Z endstop doesn't get used. I suppose I was just hoping for an integrated solution.

Surion79, 2023-05-29:
I probably would request a TMC5160 integration to start with and go from there. Set kinematic position to 'uncomplicate' things sounds for me not really like a good solution keeping functional safety into mind. Since Klipper does not provide axis-based M84, I guess you are running with it without proper homing at some point.

Frix-x, 2023-06-11:
Closing the issue as this was done and integrated into Klippain v4 that was just released :)

closed 2023-06-11 by Frix-x