Frix-x/klippain #220: Added sleep time customization (nozzle cleaning - nozzle heating)

elpopo-eng, 2023-05-31:
It's will be nide add sleep time customization.

With Bambulab Hot end and the low temperature inertia means I can wait just a few seconds instead of 30.

So we need add 2 vars : 
- sleeptime_heating
- sleeptime_oozing

elpopo-eng, 2023-05-31:
> > It's will be nide add sleep time customization.
> > With Bambulab Hot end and the low temperature inertia means I can wait just a few seconds instead of 30.
> > So we need to add 2 vars :
> > 
> > * sleeptime_heating
> > * sleeptime_oozing
> 
> I think this is not a thing to bother anymore as I fully removed this G4 in the develop branch. It was an artifact from my trials in the past and it's indeed not really needed at all IMHO

In fact I would like to put them to 0 but I don't know if someone realy need it.

elpopo-eng, 2023-06-08:
@Frix-x I've checked and ozzing sleep is always present in develop branch.
https://github.com/Frix-x/klippain/blob/1539dc4eb5f026698f48925398e910dbf5b03c54/macros/helpers/nozzle_cleaning.cfg#L113

Frix-x, 2023-06-09:
This was done in 79a86b29bbccccd93bc5c966536e4dbf851a6456

Frix-x, 2023-06-11:
Closing as Klippain v4 is merged and released :)

closed 2023-06-11 by Frix-x