Frix-x/klippain #221: Add optionnal Park in Pause

elpopo-eng, 2023-05-31:
When we use the pause, the tool head doesn't move and therefore burns the surface where it stops.

I therefore propose to add an optional park in the pause

Surion79, 2023-05-31:
Good point, need to check compatibility with ERCF and possible purge/clean before resume.

elpopo-eng, 2023-05-31:
I already do this with the ercf and there's no problem.
But I don't think it's mandatory to add a clean purge and it would be dangerous to do a clean purge when you've already printed something. You could have a collision with the printed object on the bed.

Benoitone, 2023-05-31:
In think I have put a park in pause when we rework ercf...

Frix-x, 2023-05-31:
Yes this was reworked already in the develop branch :)

elpopo-eng, 2023-06-08:
I see it and it works.
So i close this issue.

closed 2023-06-08 by elpopo-eng