Frix-x/klippain #226: Bed Tilting and z calibration on large bed issue

elpopo-eng, 2023-06-09:
I sometimes get an error with the z_calibration plug-in: `! The offset is larger than allowed: OFFSET=-2.471 MAX_DEVIATION=2.000` in the print_start procedure.

In fact, the error often occurred after the machine had idled and the motors had stopped on my voron 2.4 350mm.
As on the 350 the gantry moves a lot after the motors have stopped, the z origin is very different after QGL.

Is it possible to add a `G28` in `_TILT_CALIBRATE` just after `QUAD_GANTRY_LEVEL` and `Z_TILT_ADJUST`?

Benoitone, 2023-06-09:
That happen to me sometime. And the problem for me is that, on the G28 in the beginning, sometimes I have some cold filament on the nozzle, so when it make z calibration there is too much difference...
But I may have misidentified the cause... in any case when my nozzle is clean at first it never happens to me...

elpopo-eng, 2023-06-09:
> That happen to me sometime. And the problem for me is that, on the G28 in the beginning, sometimes I have some cold filament on the nozzle, so when it make z calibration there is too much [...]

There may be another cause, in my case I have a 2.5mm offset which is a bit large for a dirty nozzle (my nozzle is always hot when I do my print start).

Frix-x, 2023-06-09:
Elpopo is right on this. G28 Z is mandatory after a QGL/Z_TILT to reset the Z origin that can change a lot. On a normal machine we do it beofre the print to get a new origin, but when running the Z calibration plugin, it need to be done to get a new zero position on which the computed offset is applied (and that will remain stable between consecutive runs).

It was something that was implemented in this repository before the rebrand to Klippain and way before the modular START_PRINT. But unfortunately, I think it was lost during the implementation of TAP and the modular approach. As I'm using GalileoZ on my machine with virtually no gantry sag I did not catched it until today !

So I will definitely reimplement it ! But instead of doing it after the QGL, what do you think of doing it just before the CALIBRATE_Z in the "z_offset" module ? https://github.com/Frix-x/klippain/blob/fea91cdf8ee66d5331325a90e4f654dd0364d926/macros/base/start_print.cfg#L450-L454

Frix-x, 2023-06-09:
Ok, since we discussed about it in Discord, here is the summary of the discussion: **we need to put it after the QGL/Z_TILT**.

Indeed, for example, if the brush is put in the front of the machine, the offset change due to gantry sag could be multiple millimeters and could lead to potential crash or miss of the brush or some other equipements during the START_PRINT phase.

So I will put it after the tilting :)

Frix-x, 2023-06-09:
fixed in cfb29b923c50dd31da1488db2e66b215f6648da8

Frix-x, 2023-06-11:
Closing as Klippain v4 is merged and released :)

closed 2023-06-11 by Frix-x